simpleble/simpleble #10: Problem with writing to characteristic

PatrykSajdok, 2021-09-22:
I need to write bytes `55 06 (hex)` or `85 6 (dec)` to characteristic and my device should notify me about that, but nothing happens. Am I doing something wrong? I tried sending `U\x3` to the device, I also tried creating string from chars, but it is the same string as before. Everything works when using different library, but I need cross platform library.

kdewald, 2021-09-23:
@PatrykSajdok thanks for reporting this issue.

Do you have an example of the code that you're trying to test this with? Which device are you trying to connect to?

Keep me posted on this topic, I'll make sure we can figure it out.

PatrykSajdok, 2021-09-23:
@kdewald I was testing everything in example apps on Windows, I only modified write example app to not ask for content to write. The device I'm trying to write to is a wireless spirometer and when I will write correct bytes to the device it should clear all saved measurments.
`std::string contents;`
        `char a, b;`
        `a = 85;`
       ` b = 3;`
    `    contents.push_back(a);`
  `      contents.push_back(b);`
 `       if (selection >= 0 && selection < uuids.size()) {`
           ` peripheral.write_command(uuids[selection].first, uuids[selection].second, contents);`
       ` }`

kdewald, 2021-09-23:
Just in case, try using `write_request`, does that work?

PatrykSajdok, 2021-09-23:
`write_request` is throwing OperationNotSupported Exception

kdewald, 2021-09-23:
Thanks, then we should proceed with write_command.

I'll dig deeper to see if there is some error that we're not getting or a mistake in the actual code.

The function where all the magic happens is `PeripheralBase::write_command`, if you want to tinker with it.

kdewald, 2021-09-24:
@PatrykSajdok I confirm that I can replicate the issue locally. I'm not sure yet what is causing it.

kdewald, 2021-09-24:
Ok, it seems to be a buffer not getting properly populated. Will fix it shortly.

kdewald, 2021-09-24:
@PatrykSajdok It's fixed in the latest commit on `main`. Let me know if you still have issues.

PatrykSajdok, 2021-09-24:
Thank you very much, I will check that

PatrykSajdok, 2021-09-24:
Everything is working now, weirdly `write_request` and `write_command` are writing correctly to the device, but request is throwing an exception (after pressing Continue in VS it works)

kdewald, 2021-09-26:
> Everything is working now, weirdly `write_request` and `write_command` are writing correctly to the device, but request is throwing an exception (after pressing Continue in VS it works)

Can you provide mode information the error that you're getting? Maybe a screenshot of the exception being thrown?

PatrykSajdok, 2021-09-28:
It is just OperationNotSupported exception, in PeripheralBase when its validating

tomixnscale89, 2023-07-30:
Hello,

I'm trying to do the same thing and I'm still getting OperationNotSupported. If I use this code:

```
std::string command;

    char a, b, c;
    a = 0x00;
    b = 0x46;
    c = 0x02;

    command.push_back(a);
    command.push_back(b);
    command.push_back(c);

    peripheral.write_command(uuids[selection.value()].first, uuids[selection.value()].second, command);
```
The device will respond, but only after I click continue after the exception. Any ideas how to continue forward?