simpleble/simpleble #9: Cannot subscribe to characteristic

PatrykSajdok, 2021-09-22:
I'm trying to read data from few bluetooth devices, for most devices everything works, but for one I cannot subscribe to correct characteristic, because `notify` method in `PeripheralBase` is throwing OperationNotSupported Exception, which is weird because other library I tried was able to read data from the same characteristic. I'm using `notify` example from examples.

PatrykSajdok, 2021-09-22:
Update: looks like `indicate` method works just fine

closed 2021-09-22 by PatrykSajdok

kdewald, 2021-09-23:
@PatrykSajdok thanks for reporting this issue.

You are correct, if you get OperationNotSupported when calling `notify`, it means that the characteristic will probably only support indications. Other library/OS implementations treat both as if they were the same, but for the sake of transparency to the user I try to enforce the right function to be used.