simpleble/simpleble #11: [Proposal, Feature Request] Add void * to all callback in plain C API

Andrey1994, 2021-09-23:
I want to use plain C API in C++ code, cannot use C++ API because I don't want to link SimpleBLE since it requires system-level packages.

To make it easier it will be good to add `void *` to all plain C callbacks as an argument. (It's like in the `CreateThread` method from WinAPI, there is a way to pass one parameter as `LPVOID`). With such an argument it will be possible to write smth like:

```
// class with some methods and real callback
class MyClass {
public:
    void adapter_on_scan_found (simpleble_adapter_t adapter, simpleble_peripheral_t peripheral);
};
....
// plain C callback for SimpleBLE C library
void adapter_on_scan_found(simpleble_adapter_t adapter, simpleble_peripheral_t peripheral, void *obj)
{
    ((MyClass *)(obj))->adapter_on_scan_found(adapter, peripheral);
}
....
// in another class method will be possible to write smth like that
simpleble_adapter_set_callback_on_scan_found (adapter, adapter_on_scan_found, (void *) this);
```

kdewald, 2021-09-23:
Excellent feature request. I'll add that asap.

kdewald, 2021-09-23:
Added!

closed 2021-09-23 by kdewald