simpleble/simpleble #13: Memory error free `simpleble_peripheral_address`

Andrey1994, 2021-09-23:
I am not 100% sure and cannot find the exact issue but when I run smth very similar to connect_c sample with the debugger in VS I get this assert. I didn't find a clue and I don't see the same issue for `simple_peripheral_identifier`

[image]

Andrey1994, 2021-09-24:
[image]

Andrey1994, 2021-09-24:
I found a hint but dont know the best solution for it... Take a look at this answer https://stackoverflow.com/a/35311928 

I am linking MSVC runtime statically because this way users who use precompiled libs(majority of users in fact) don't need to install C++ redist package.

Potential solutions:

* stop using static linking of MSVC runtime(not an option in fact, I don't want to force users to install extra packages, especially when before they didn't have to)
* As this guy proposed "You could get that work even with /MT, but that is more difficult - you would need to provide some interface which will allow the objects allocated in the DLL to be deallocated there as well. " The question is should I do it only in my project or it should be in SimpleBLE upstream?

Andrey1994, 2021-09-24:
Or maybe it would be better to change API to allocate memory outside SimpleBLE dll
Now: 
```
char * simpleble_peripheral_address (simpleble_peripheral_t handle)
```
After this change:
```
size_t simpleble_peripheral_address (simpleble_peripheral_t handle, char *address_output, size_t max_size) // return actual size
```

And do the same for all other methods

kdewald, 2021-09-24:
I think it might be better to add a `simpleble_free` method. I'll add that asap.

kdewald, 2021-09-24:
I just added the `simpleble_free` method, let me know if you see any additional issues.

closed 2021-09-24 by kdewald

Andrey1994, 2021-09-24:
it helped, now it works fine, thanks