simpleble/simpleble #21: `winrt::init_apartment();` fails when its called from Python on Windows

Andrey1994, 2021-09-27:
Hi, seems like you are going to add python binding for this library, so more likely you will face exactly the same issue or have already seen it.

When I call it from C\C++, everything is fine. In Python I call it using ctypes and `LoadLibrary`(I load dll from BrainFlow and BrainFlow dll loads simpleble-c.dll in runtime).

This method fails `winrt::init_apartment();` full stacktrace is attached.

[image]

I am debugging and googling it right now, any ideas are welcome

Andrey1994, 2021-09-27:
Steps to reproduce using BrainFlow(in case if you think its faster than replicating the issue from the scratch):

* clone https://github.com/brainflow-dev/brainflow
* `git checkout ble`
* `python tools\build.py --ble --debug` (or manually type underlying cmake commands)
* `cd python-package`
* `python -m pip install -U .` 
* open this file https://github.com/brainflow-dev/brainflow/blob/ble/tests/python/brainflow_get_data.py and add `input()` somewhere before `board.prepare_session()` Its needed to have time to connect to python process with a debugger
* `python tests\python\brainflow_get_data.py --board-id 39`
* open visual studio as admin -> debug -> attach to process -> python.exe
* press smth in a terminal and you will see the error

Using C\C++ tests I don't see any problems with it

kdewald, 2021-09-28:
Also related to this one: https://github.com/microsoft/cppwinrt/issues/53

kdewald, 2021-09-28:
Extra information: https://stackoverflow.com/questions/16466641/how-to-declare-and-link-to-roinitialize-rouninitialize-rogetactivationfactory-an

PatrykSajdok, 2021-09-28:
I have the same problem when I tried to use this library in Unity, I've added more methods like CacheAdapter, StartScan (on cached adapted), etc. and everything is working in regular C# console application, but in Unity it cannot use `winrt::init_apartment();`

PatrykSajdok, 2021-09-29:
@kdewald In Unity it shows this error:
[image]
but strangely, after pressing Ignore 3 times everything is working

PatrykSajdok, 2021-09-29:
@kdewald Update: When I run all bluetooth calls in Task instead of main thread everything is working, you are the best!

Andrey1994, 2021-09-29:
It was my guess to move all SimpleBLE API calls into its own thread, but for me, it would require much more work to do and I would like to avoid it

kdewald, 2021-09-30:
I agree that running all calls from a different thread is not a solution, so we'll keep trying until we figure it out.

kdewald, 2021-09-30:
@Andrey1994 you mentioned on the PR it failed on "the next line", was it:

`winrt::hresult const result = WINRT_RoInitialize(static_cast<uint32_t>(type));`

or

`auto device_selector = BluetoothAdapter::GetDeviceSelector();`

kdewald, 2021-09-30:
@Andrey1994 I've tried your example with the latest version of the `bugfix/winrt` branch and it doesn't seem to crash.

Andrey1994, 2021-09-30:
More likely you've made a mistake somewhere in the installation process.

Check that you've changed the file in third_party\SimpleBLE and installed python package from a local folder instead PYPI

About exact line: I think there are multiple lines where it fails(I see such window 3-5 times) if I press ignore. Here is the first one 
[image]

Andrey1994, 2021-09-30:
Or maybe you've compiled it without `--debug` flag and for release builds these asserts are probably disabled...

kdewald, 2021-09-30:
What happens if you remove the WinRT initialization altogether, does the code work or do you still see errors?

kdewald, 2021-10-01:
I attempted to run your instructions again, this is the result I got in the terminal, but no assertions or hard faults.

[image]

Andrey1994, 2021-10-01:
For some reason for you it fails earlier with simpleble-c.dll loading.

Could you check that this file exists at python-package/brainflow/lib folder after compilation?

And after pip install it should be stored at site packages folder

Andrey1994, 2021-10-01:
Also, I do not think, that its the only problem calling SimpleBLE from Python or smth else(seems like for Unity its the same). If you take a look at the latest screenshot you will notice that now it fails(debug build) with assert like `!is_sta()` thread

kdewald, 2021-10-02:
According to this post https://stackoverflow.com/questions/51487406/how-can-i-get-the-chosen-file-from-a-fileopenpicker-window-in-c-winrt that exception was because WinRT was initialized in multi-threaded mode.

kdewald, 2021-10-02:
I just updated https://github.com/OpenBluetoothToolbox/SimpleBLE/tree/bugfix/winrt with what I think should be the solution.

Andrey1994, 2021-10-04:
the original problem is gone, now with debug builds, I see plenty of asserts triggered like `!is_sta()`(even for c++ samples) which come from winrt, as long as it works for release builds without any notable issues I don't care ***too*** much about them.

If I realize that its a problem I will try to move it all into its own thread

closed 2021-10-04 by Andrey1994