simpleble/simpleble #24: Sending 0 byte

PatrykSajdok, 2021-10-04:
I'm not sure if this is because I'm using this library in C#, but I had a problem where I had to send specific bytes to the device and in these bytes there was 0. When I was sending this, it was not detecting that 0 and was just ignoring it. I found a workaround for that by sending bytes using `char*` and passing array lenght. If someone will have similar problem this may be helpfull.

kdewald, 2021-10-04:
How are you creating the ByteArray object? Remember that it is a wrapper around `std::string`, which requires some specific handling to ensure it recognizes special characters.

If you share some code example I can show you how to do this correctly.

PatrykSajdok, 2021-10-04:
As I cannot receive string from C#, I'm receiving all data as `char*` and I was just passing it to `write` methods. I'm not really a C++ developer so I didn't know that I shouldn't do it like that. My code is different now, but I think it was looking like this:
header:
`SIMPLEBLEDLL_API void WriteToCharacteristicCommand(char* serviceUUID, char* characteristicUUID, char* bytes);`
source:
`void WriteToCharacteristicCommand(char* serviceUUID, char* characteristicUUID, char* bytes) {
    CachedDevice.write_request(serviceUUID, characteristicUUID, bytes);
}`

kdewald, 2021-10-04:
Yeah, that code will have the issue you mention. You should probably use the C wrapper for the library instead, that will take care of those conversions for you and in my opinion is better suited for that you're trying to achieve. If that doesn't fully meet your needs, make sure you read through https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/main/src_c/peripheral.cpp as it's basically doing the same thing you are doing.

Let me know if you have any questions.

kdewald, 2021-10-04:
PS: To initialize a `SimpleBLE::ByteArray`, you should always use this example: `SimpleBLE::ByteArray((const char*)data, data_length)`

closed 2021-10-04 by kdewald

PatrykSajdok, 2021-10-04:
Thank you very much, I will check this