simpleble/simpleble #31: Windows 11 python deadlock

Andrey1994, 2021-11-04:
After updating from Windows 10 to Windows 11 in Python I get a deadlock(never ends, 0% CPU usage). Its weird that I cannot reproduce it in C++

Connected with a debugger to python and got that call stack:

[image]

In attempt_connect `async_get` never ends due to some reason.

Not a solution for this particular issue but maybe it makes sense to add an option to configure the timeout for waiting? Non-zero exit code or exception is definitely better than deadlock

Andrey1994, 2021-11-04:
If I replace `async_get` by `.get()` it goes further until next `asyn_get` call 
[image]

Andrey1994, 2021-11-04:
ah, no, it seems like it was random and happened once, `get()` also hangs

kdewald, 2021-11-04:
This could also be at Win11 thing. Does it connect at all? (Even after changing the timeout?)

Andrey1994, 2021-11-04:
it connects using c++ samples, using python somehow I've managed to run it once but it was random. It looks similar to this https://stackoverflow.com/questions/55594814/bug-in-windows-bluetooth-le-getgattservicesasync-method but it is almost for all calls and fails in 99%.

Andrey1994, 2021-12-03:
A relatively simple way to reproduce: replace this `if (strncmp (peripheral_identified, "Muse", 4) == 0)` at https://github.com/brainflow-dev/brainflow/blob/master/src/board_controller/muse/muse.cpp#L426 to match the name of the device which you have, rebuild brainflow, reinstall python binding, and run test for muse

```
python tools\build.py --ble
cd python-package
python -m pip install -U .
python ../tests/python/brainflow_get_data.py --board-id 38
```

Andrey1994, 2021-12-23:
Also, any hints or ideas why it works for c++ and does not work if I call the same code from python can be helpful. And why its different for win10 and 11

kdewald, 2021-12-23:
I suspect there is something different in how threading is handled by a standalone program vs python. So far I haven't found anything useful, but I also admit that my focus has been on the Linux rewriting (which hopefully is close to be ready) and I can promise that my next focus will be to get this working on Windows 11 and other environments too.

kdewald, 2021-12-30:
@Andrey1994 can you tell me which version of the Windows SDK you are using to build your Windows targets?

I just realized there is a new Windows SDK that came out that is referencing Windows 11 (see attached image), so I'm wondering if something changed here that could be causing problems, potentially by running on Windows 11 the library compiled on Windows 10 or some older version of the SDK.

I'll keep on investigating to see what I can find before upgrading my own setup, as I'll have no way of rolling back once I do.

[image]

Andrey1994, 2021-12-30:
good idea, I will try the newest one
I am still using 10.0.19041.0 for simpleble

kdewald, 2021-12-30:
I just put up a PR with the necessary changes to compile with the latest one: https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/39

Andrey1994, 2021-12-30:
tried with 10.0.22 and v143, unfortunately it didn't help

kdewald, 2021-12-30:
Just to confirm: Are you building the library on Windows 10 or 11?
Which version of Visual Studio are you using?

Andrey1994, 2021-12-30:
On win11 locally using VS 2022, I use CMake with different options for the toolchain, winsdk version, etc. I see exactly the same behavior for all of them. In CI I use older VS

kdewald, 2021-12-31:
@Andrey1994 I just noticed another thing that might be related:

On `brainflow/tools/build.py`, you have this line here:
`parser.add_argument('--cmake-system-version', type=str, help='system version for win', required=False, default='8.1')`

This is causing the CMake/Visual Studio to generate this option:
`-- Selecting Windows SDK version 10.0.22000.0 to target Windows 8.1.`

I'm wondering if this is causing the problem you're seeing, as this setting will most likely be propagated down to SimpleBLE as well. I'll test this to see if some weird behavior does happen when this setting is specified.

Andrey1994, 2022-01-01:
I also thought about it, and I don't fully understand how it works since simpleble requires 10.0.19+.
But it was fine for win10.
In CI pipelines I do it differently, I am building SimpleBLE outside from BrainFlow manually using -DCMAKE_SYSTEM_VERSION 10.0.19 and build BrainFlow independently from it and copy-paste all DLLs after that. (https://github.com/brainflow-dev/brainflow/blob/master/.github/workflows/run_windows.yml#L49) And both options worked fine for win10

Also, I've updated it locally to add vs22 and fix cmake-system-version default but have not pushed it yet, so I've tested it with different values

kdewald, 2022-01-01:
Great!

Let me repeat what you said to make sure I understood everything correctly: When changing `CMAKE_SYSTEM_VERSION` to 10.0.19+ on `build.py`, the issue you were describing in Windows 11 went away.

Is this correct? If so, can I close this issue or is there anything else you want to draw my attention to? (I have a few more things I would like to fix on the Windows build, which are unrelated.)

Andrey1994, 2022-01-01:
No, not at all.
I said that this issue exists on win11 no matter which windows SDK version I specify. And do not exist on win10 even if I specify 8.1

kdewald, 2022-01-01:
Thank's for confirming. I'll keep digging deeper then.

kdewald, 2022-01-03:
@Andrey1994 I've followed your example above on a fairly clean installation of Windows 11 and everything worked fine.

[image]

I'm starting to wonder if there is some other component that might be affecting here, like the choice of BLE dongle or some other software component that might be breaking this. Have you tried running the `connect` or `notify` examples on a Muse device to see if that part is working as expected?

Also, I found that sometimes there is an OS service related to Bluetooth that needs to be manually restarted as it might not migrate nicely to Windows 11 ([link here](https://appuals.com/bluetooth-not-working-windows-11/)):
[image]

Andrey1994, 2022-01-03:
Interesting,

<< Have you tried running the connect or notify examples on a Muse device to see if that part is working as expected?
It works fine if I call it from C\C++ binding for BrainFlow

I think ble service works properly since it works if I call it from C++ API directly, I've also tried to restart this service and tested it for a few more devices(not Muse) and it always hangs at getting services of the peripheral

Andrey1994, 2022-01-03:
Did you make any changes in code to run it? Smth like winsdk version, etc

kdewald, 2022-01-04:
> Did you make any changes in code to run it? Smth like winsdk version, etc

None, I cloned for `master` and that was it.

One idea I had was to see if we can replicate this problem using Windows Sandbox, which should allow to test this with a  "clean" installation.

Also, if you test the latest commit from `main`, I did add a 10 second timeout to the `async_get` function, so maybe that can provide some additional feedback of what might be going on.

Andrey1994, 2022-01-06:
I've tried with updated simpleble and indeed now it ends after 10 seconds. But I don't get any logs\messages, it just ends. I will try to get more info with a debugger but don't expect anything new.

Also, I thought to try with an external BLE dongle(BLED112 from silicon labs) but SimpleBLE(or Windows?) doesn't recognize it as BLE adapter. E.g. `simpleble_adapter_get_count ();` returns o

Andrey1994, 2022-01-06:
the problem is the same, but seems like `async.GetResults();` throws unhandled exception in case if timeout triggered

[image]

Andrey1994, 2022-01-06:
Should I be able to connect to the device using Bluetooth settings? Everything works fine if I run it from console app using C\C++ API

[image]

kdewald, 2022-01-06:
> I've tried with updated simpleble and indeed now it ends after 10 seconds. But I don't get any logs\messages, it just ends. I will try to get more info with a debugger but don't expect anything new.

That means the timeout was reached and an OS exception is generated. I'll make a note to capture this one.

> Also, I thought to try with an external BLE dongle(BLED112 from silicon labs) but SimpleBLE(or Windows?) doesn't recognize it as BLE adapter. E.g. `simpleble_adapter_get_count ();` returns o

This is somewhat known, Windows has a complicated system to enumerate BLE devices and I'm pretty sure it will sometimes fail. I will have to look deeper into that at some point.

kdewald, 2022-01-06:
> the problem is the same, but seems like `async.GetResults();` throws unhandled exception in case if timeout triggered
> 
> [image]

Ok, that means that the operation is not complete (and needs some fixing). The weird thing is that this process takes so long.

kdewald, 2022-07-01:
Considering it has been 6 months since this issue was last updated and new exception handling will be part of v0.5.0. I think we can close this issue. Feel free to open a new one if problems arise.

Andrey1994, 2022-07-01:
Actually it still doesnt work for me and I have no ideas how to investigate it further. And I am still not sure if its only for my laptop or not

kdewald, 2022-07-09:
So far I haven't been able to replicate it and no other reports have come in about this. I suspect there might be some weird deadlock or concurrency issue that could be affecting, but there are plenty of additional checks right now to prevent this stuff from happening. Are you using the current commit at the tip of `main`? That one should be the most stable version.

Andrey1994, 2022-07-09:
Using the latest commit instead deadlock and winrt exception I see that it fails to connect and its definitely better. 

But I reproduce it 100% times and in exactly the same place and only when I call it from python. It works for all other languages.

Andrey1994, 2022-07-09:
I think nobody else reported it probably because more likely not many people tried to call it from python and using win11..

Andrey1994, 2022-07-09:
btw it connects using python_simpleble using connect.py example

kdewald, 2022-07-09:
Can you link again the latest version of the code where this is happening? I think we have some multi-threading problems happening.

Andrey1994, 2022-07-09:
you will need a branch from this PR https://github.com/brainflow-dev/brainflow/pull/510 it has updated simpleble

Full code path is:

* Python calls C library using ctypes.cdll  https://github.com/brainflow-dev/brainflow/blob/multiple_buffers/python_package/brainflow/board_shim.py#L153
* methods in this C library have lock guards and are not executed simultaneously https://github.com/brainflow-dev/brainflow/blob/multiple_buffers/src/board_controller/board_controller.cpp#L281
* Plain C methods call methods from cpp objects, I dont link simpleble and created a helper board https://github.com/brainflow-dev/brainflow/blob/multiple_buffers/src/board_controller/inc/ble_lib_board.h it finds function addresses after manual call of LoadLibrary https://github.com/brainflow-dev/brainflow/blob/multiple_buffers/src/board_controller/ble_lib_board.cpp#L57 and I also added lock guards to these methods because I have no idea if simpleble is thread safe or not https://github.com/brainflow-dev/brainflow/blob/multiple_buffers/src/board_controller/ble_lib_board.cpp#L109 
* and it finally goes to the actual device class https://github.com/brainflow-dev/brainflow/blob/multiple_buffers/src/board_controller/muse/muse.cpp 

Inside this board class it does this : 

```c++
    simpleble_adapter_set_callback_on_scan_found (
        muse_adapter, ::adapter_on_scan_found, (void *)this);

    simpleble_adapter_scan_start (muse_adapter);
    int res = (int)BrainFlowExitCodes::STATUS_OK;
    std::unique_lock<std::mutex> lk (m);
    auto sec = std::chrono::seconds (1);
    if (cv.wait_for (lk, params.timeout * sec, [this] { return this->muse_peripheral != NULL; }))
    {
        safe_logger (spdlog::level::info, "Found Muse device");
    }
    else
    {
        safe_logger (spdlog::level::err, "Failed to find Muse Device");
        res = (int)BrainFlowExitCodes::BOARD_NOT_READY_ERROR;
    }
    simpleble_adapter_scan_stop (muse_adapter);
    if (res == (int)BrainFlowExitCodes::STATUS_OK)
    {
        if (simpleble_peripheral_connect (muse_peripheral) == SIMPLEBLE_SUCCESS)
        {
            safe_logger (spdlog::level::info, "Connected to Muse Device");
        }
        else
        {
            safe_logger (spdlog::level::err, "Failed to connect to Muse Device");
            res = (int)BrainFlowExitCodes::BOARD_NOT_READY_ERROR;
        }
    }
```

I use class methods as callbacks and its done by passing `this` to plain C callbacks. Also I use condition variable to notify main thread if required device is found to dont wait more time than needed

Andrey1994, 2022-07-09:
simpleble_peripheral_connect fails with timeout in _attempt_connect and async_get

kdewald, 2022-07-11:
Can you print the output that you get from `src/windows/Utils.cpp` in the `initialize_winrt` function? I suspect there's an initialization error there.

kdewald, 2022-07-11:
Also, in `Utils.h` you should check what error is returned in the catch statements of `async_get`, that could also give a clue.

Andrey1994, 2022-07-11:
@kdewald thanks for the hint. 
I added:

```cpp
     WINRT_IMPL_CoGetApartmentType(&cotype, &qualifier);

+    std::cerr << "cotype is " << cotype << "qualifier " << qualifier << std::endl;
+
     if (cotype == -1 /* APTTYPE_CURRENT */) {
         // TODO: Investigate if multi or single threaded initialization is needed.
         winrt::apartment_type const type = winrt::apartment_type::multi_threaded;
         winrt::hresult const result = WINRT_IMPL_CoInitializeEx(nullptr, static_cast<uint32_t>(type));
+        std::cerr << (long)result << std::endl;
     }
```

And:

```cpp
        return async.GetResults();
     } catch (const winrt::hresult_error& err) {
+        std::cerr << (long)err.code().value << " " << winrt::to_string(err.message()) << std::endl;
         throw SimpleBLE::Exception::WinRTException(err.code().value, winrt::to_string(err.message()));
     } catch (const std::exception& err) {
+        std::cerr << err.what() << std::endl;
```

The output is:

```
cotype is 3qualifier 0
-2147483634 A method was called at an unexpected time. // this one from attempt_connect
```

So, WINRT_IMPL_CoInitializeEx is not even called

Andrey1994, 2022-07-11:
for other languages cotype is -1 and its executed. But I have not checked other languages with GIL...

kdewald, 2022-07-13:
I think you need to call that function from within your library in the main thread. I'll keep looking to see if anything else comes up.

kdewald, 2022-07-13:
Also, what Python version are you using? That could also be related.

Andrey1994, 2022-07-13:
its in the main thread. Since simpleble creates its own threads for callbacks,etc I dont need to create any other threads manually
python 3.9.6 x64

Andrey1994, 2022-07-13:
wow, it works if I use 3.10.5! Thanks a lot I was struggling with it for months!

eriklins, 2023-09-01:
Hey all, seems I'm running into the same issue now with my [InsideBlue](https://github.com/eriklins/InsideBlue-BLE-Tool) tool, which use my Pascal bindings. Got a new laptop running Windows 11 and I can no longer connect to any BLE device from InsideBlue. Scanning works as usual but connecting attempt takes long and finally fails everytime. I recompiled the SimpleBLE DLLs with the latest pull from github on Windows 11, but no change. Since the issue has been close quite a while back, what did fix the issue with Python? Any other clues?