simpleble/simpleble #33: macOS Monterey examples not working

fidoriel, 2021-11-30:
On macOS the examples do compile and run but will not find any BLE device. Same version of code is running on Linux.

Reproduce:
execute e.g. scan sample from console or from finder on macOS (M1 and Intel not woking) => no device will be detected. 

A desktop App using the lib is working. Bluetooth permissions are granted in settings for iTerm.

microbuilder, 2021-12-12:
This may be a bug on Apple's part. Here is another project with the same problem: https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/33

Modifying the scan function in SimpleBLE, if instead of scanning for any device, you set it to scan for a specific service then it works ok.

Hoepfully the next OS update from Apple will fix this?

kdewald, 2021-12-13:
Thanks for the update, that's good information. However my suspicion is that this has to do with permissions. If you give permissions to the Terminal app, there's a chance this might work again.

I haven't yet had the chance to update to Monterrey, but I'll keep a close eye on this.

microbuilder, 2021-12-13:
> Thanks for the update, that's good information. However my suspicion is that this has to do with permissions. If you give permissions to the Terminal app, there's a chance this might work again.
> 
> I haven't yet had the chance to update to Monterrey, but I'll keep a close eye on this.

I gave Bluetooth permissions to the terminal app, and as mentioned scanning for a specific service does work, so I don't think it's a permission issue in this case. Apparently there is an OS update coming this week from Apple ... I'll test it out once I apply the update.

microbuilder, 2021-12-14:
I just updated to 12.1, and, alas, scanning issues persist:

```bash
$ sw_vers
ProductName:	macOS
ProductVersion:	12.1
BuildVersion:	21C52
```

```bash
$ ./example_scan_c 
Adapter Default Adapter started scanning.
Adapter Default Adapter stopped scanning.
```

```bash
$ ./example_scan  
Scan started.
Scan stopped.
Scan complete.
The following devices were found:
```

kdewald, 2021-12-19:
Thanks for the report! There is definitely something related to permissions that is messing here. I'll keep researching.

kdewald, 2022-01-06:
@fidoriel @microbuilder quick update on the topic: It seems that whatever Apple changed messed something up with the OS permissions, but I do have a few ideas I will try to pursue in the upcoming days to see if this can be fixed. I'll keep you posted.

fidoriel, 2022-01-12:
In the mean time this bug? occurred on my GUI (macOS 12.1) too. I was able to solve this by singing the binary, but the CLI is still broken. @kdewald Thank you for your effort

kdewald, 2022-01-19:
I'm still looking into this. I also tested the beta of macOS 12.2 and it also seems to be broken. It clearly seems like it's something permission-related, but it's unclear now how to fix it.

kdewald, 2022-01-19:
I just found this: https://github.com/hbldh/bleak/issues/720#issuecomment-1009198158

microbuilder, 2022-01-19:
> I just found this: https://github.com/hbldh/bleak/issues/720#issuecomment-1009198158

That makes sense, but also seems problematic for a command line tool if we need a signed app running in the foreground (ie a GUI app) to scan for arbitrary devices. 😣 Good for Apple taking security seriously with things like this, but I think scanning with a command line tool in a foreground terminal instance is a perfectly valid use case as well.

kdewald, 2022-01-20:
I think this is caused by Apple trying to have the same power-saving features on MacOS as iOS has, thus forcing applications to be very explicit about what resources they need to keep running.

I've been playing a lot with the build process and might have a few leads on how to work around this issue in a realistic manner. I don't want to make any promises yet in case my ideas are dumb, but at least there's hope :P

nate2427, 2022-01-20:
> Modifying the scan function in SimpleBLE, if instead of scanning for any device, you set it to scan for a specific service then it works ok.

@microbuilder How would I get it to scan for the Muse S device? Im looking into the code at AdapterBase.mm, but I am a little confused here.

kdewald, 2022-01-20:
@nate2427 if you want to hack the library just to scan for that device, this is the line you need to modify, by specifying the specific Muse service: https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/f9ec8ea9be351f06872585bc0abbd1feca1053d0/src/macos/AdapterBaseMacOS.mm#L54

nate2427, 2022-01-20:
Thanks @kdewald ! I was able to find that line earlier, but I actually dont know how to update it to look for the Muse S. Are you aware of a document that describes how to manipulate this line for Muse S specifically?

kdewald, 2022-01-20:
> Thanks @kdewald ! I was able to find that line earlier, but I actually dont know how to update it to look for the Muse S. Are you aware of a document that describes how to manipulate this line for Muse S specifically?

I'm not. SimpleBLE is designed to be generic, so I'm not familiar with any device specifics.

nate2427, 2022-01-20:
@kdewald Thanks man! I'll continue researching and update the post whenever I get it to work.

nate2427, 2022-01-20:
@kdewald @microbuilder @fidoriel I was able to find the services the Muse S exposes and updated the scan function as follows:

`- (void)scanStart {
    [self validateCentralManagerState];
    [self.centralManager scanForPeripheralsWithServices:@[CBUUID UUIDWithString:@"FE8D"] options:@{CBCentralManagerScanOptionAllowDuplicatesKey : @YES}];
    if (!self.centralManager.isScanning) {
        NSLog(@"Did not start scanning!\n");
        // TODO: Should an exception be thrown here?
    }
}`

After rebuilding and running the exam, I am still unable to get my Muse S band connected. Any suggestions why? Im not the strongest in objective-c syntax.

nate2427, 2022-01-20:
@fidoriel you mentioned signing earlier, what exactly does that entail?

microbuilder, 2022-01-20:
@nate2427 This is really somewhat of a separate issue ... I'd suggest filing it as a new issue, just to track the two different use cases separately? This one should cover fixing scanning for every device with the recent changes to OS X, whereas how to scan for a single device is a related by separate problem to document or explore. Feel free to tag the same people, and it's just a suggestion, but it's usually good practice to keep issues and PRs narrowly focused.

nate2427, 2022-01-20:
Great point. Really my first time commenting on issues. I appreciate the knowledge drop fr 🙏🏾

kdewald, 2022-01-23:
Ok, after a lot of trial and error, I was able to get some logs out of bluetoothd on Monterey:

```
default	23:46:30.937401-0800	bluetoothd	PowerManager power state is 1
default	23:46:30.940047-0800	bluetoothd	PowerManager power state is 1
default	23:46:30.940449-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	23:46:34.761101-0800	bluetoothd	[MacUserClient][NotificationEventHandler] -- Received Heart Beat notification -- HIDShim
default	23:46:35.936872-0800	bluetoothd	PowerManager power state is 1
default	23:46:35.937011-0800	bluetoothd	PowerManager power state is 1
default	23:46:35.937459-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	23:46:42.811285-0800	bluetoothd	PowerManager power state is 1
default	23:46:42.811532-0800	bluetoothd	PowerManager power state is 1
default	23:46:42.812177-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	23:46:45.936685-0800	bluetoothd	PowerManager power state is 1
default	23:46:45.936805-0800	bluetoothd	PowerManager power state is 1
default	23:46:45.937202-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	23:46:50.936599-0800	bluetoothd	PowerManager power state is 1
default	23:46:50.936713-0800	bluetoothd	PowerManager power state is 1
default	23:46:50.937108-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	23:46:51.549931-0800	tccd	AUTHREQ_CTX: msgID=2060.1, function=<private>, service=kTCCServiceBluetoothAlways, preflight=yes, query=1,
default	23:46:51.590960-0800	example_scan	Application does have Bluetooth authorization.
default	23:46:51.592315-0800	bluetoothd	PowerManager power state is 1
default	23:46:51.592449-0800	bluetoothd	Received XPC message "CBMsgIdCheckIn" from session ""
default	23:46:51.592832-0800	tccd	AUTHREQ_CTX: msgID=2060.2, function=<private>, service=kTCCServiceBluetoothAlways, preflight=yes, query=1,
default	23:46:51.592535-0800	bluetoothd	Received XPC check-in from session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37" fAccessLevel 0 fProgrammaticPairing 0
default	23:46:51.593437-0800	bluetoothd	Access level is less than kXPCAccessLevelSystem for session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37". Restricted state operation not allowed
default	23:46:51.593496-0800	bluetoothd	Sending 'session attached' event for session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.593713-0800	bluetoothd	Attached session for "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37" with session: 0x7fce2b035580, session handle: 0xc1370000
default	23:46:51.593761-0800	bluetoothd	Registering central session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37" with backgrounding: off, persistence: off (CBSR) restoreID: (null)
default	23:46:51.594599-0800	runningboardd	[daemon<com.apple.bluetoothd>:740] handle lookup could not find a matching process
default	23:46:51.595171-0800	runningboardd	[daemon<com.apple.bluetoothd>:740] handle lookup could not find a matching process
default	23:46:51.595490-0800	bluetoothd	Session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37" tccRequired : 1
default	23:46:51.595541-0800	bluetoothd	ReadyForTCC. TCC required : 1
default	23:46:51.595601-0800	bluetoothd	Sending XPC message "CBMsgIdReadyForTCC" to session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.595647-0800	bluetoothd	Session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37" : needsRestrictedStateOperation = 0, overrideRestrictedState = 0 , denylistMode = 0, receivesControllerBTClockEvents=0
default	23:46:51.595676-0800	bluetoothd	PowerManager power state is 1
default	23:46:51.595979-0800	example_scan	Received CBMsgIdReadyForTCC
default	23:46:51.596016-0800	example_scan	Running performTccCheck CBManager tccAvail 1, tccRequired 1
default	23:46:51.596041-0800	example_scan	TCC required
default	23:46:51.597168-0800	tccd	AUTHREQ_CTX: msgID=2060.3, function=<private>, service=kTCCServiceBluetoothAlways, preflight=no, query=1,
default	23:46:51.628277-0800	example_scan	Application does have Bluetooth authorization.
default	23:46:51.631831-0800	bluetoothd	Received XPC message "CBMsgIdTCCDone" from session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.631911-0800	bluetoothd	PowerManager power state is 1
default	23:46:51.631936-0800	bluetoothd	Power state changed. PowerState:1 managerState:5 fUserAuthorization: 0	 medicalDeviceorStateRestorationAppPendingTCC:0
default	23:46:51.631952-0800	bluetoothd	Sending 'state updated' event with state "On" to session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.631967-0800	bluetoothd	Sending XPC message "CBMsgIdStateUpdated" to session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.631995-0800	bluetoothd	Sending XPC message "CBMsgIdSupportedFeatures" to session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.632030-0800	bluetoothd	advertisingAddressDidChange connectable:<private> non-connectable:<private>
default	23:46:51.632669-0800	bluetoothd	Ignore advertisingAddressDidChange for session <private> with access level : 0 and manager state not powered on
[... cut ...]
default	23:46:57.683969-0800	bluetoothd	Erasing session 0x7fce2b035580 from SessionMap for "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:47:01.511583-0800	bluetoothd	PowerManager power state is 1
default	23:47:01.511652-0800	bluetoothd	PowerManager power state is 1
default	23:47:01.512249-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	23:47:04.761612-0800	bluetoothd	[MacUserClient][NotificationEventHandler] -- Received Heart Beat notification -- HIDShim
```

kdewald, 2022-01-23:
And these are the logs when scanning using Bluetility, which is still working:

```
default	09:30:00.940453-0800	bluetoothd	PowerManager power state is 1
default	09:30:00.940567-0800	bluetoothd	PowerManager power state is 1
default	09:30:00.940973-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	09:30:04.142606-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.142841-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.143399-0800	bluetoothd	Posting Bluetooth Status Notification <private>
default	09:30:04.218543-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.218626-0800	bluetoothd	Received XPC message "CBMsgIdCheckIn" from session ""
default	09:30:04.218672-0800	bluetoothd	Received XPC check-in from session "com.rossible.Bluetility-central-2234-40" fAccessLevel 0 fProgrammaticPairing 0
default	09:30:04.218693-0800	bluetoothd	Access level is less than kXPCAccessLevelSystem for session "com.rossible.Bluetility-central-2234-40". Restricted state operation not allowed
default	09:30:04.218729-0800	bluetoothd	Sending 'session attached' event for session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.218754-0800	bluetoothd	Attached session for "com.rossible.Bluetility-central-2234-40" with session: 0x7fce2b0168f0, session handle: 0xb09c0000
default	09:30:04.218780-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.218808-0800	bluetoothd	Registering central session "com.rossible.Bluetility-central-2234-40" with backgrounding: off, persistence: off (CBSR) restoreID: (null)
default	09:30:04.219702-0800	bluetoothd	<private> == <private>
default	09:30:04.220556-0800	bluetoothd	<private> == <private>
default	09:30:04.220592-0800	bluetoothd	Session "com.rossible.Bluetility-central-2234-40" tccRequired : 1
default	09:30:04.220618-0800	bluetoothd	ReadyForTCC. TCC required : 1
default	09:30:04.220645-0800	bluetoothd	Sending XPC message "CBMsgIdReadyForTCC" to session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.220693-0800	bluetoothd	Session "com.rossible.Bluetility-central-2234-40" : needsRestrictedStateOperation = 0, overrideRestrictedState = 0 , denylistMode = 0, receivesControllerBTClockEvents=0
default	09:30:04.220735-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.376688-0800	Bluetility	Received CBMsgIdReadyForTCC
default	09:30:04.376738-0800	Bluetility	Running performTccCheck CBManager tccAvail 1, tccRequired 1
default	09:30:04.376761-0800	Bluetility	TCC required
default	09:30:04.377628-0800	tccd	AUTHREQ_CTX: msgID=2234.2, function=<private>, service=kTCCServiceBluetoothAlways, preflight=no, query=1,
default	09:30:04.408869-0800	bluetoothd	Received XPC message "CBMsgIdTCCDone" from session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.410125-0800	tccd	AUTHREQ_CTX: msgID=740.4, function=<private>, service=kTCCServiceBluetoothAlways, preflight=yes, query=1,
default	09:30:04.410171-0800	tccd	AUTHREQ_ATTRIBUTION: msgID=740.4, attribution={accessing={<TCCDProcess: identifier=com.rossible.Bluetility, pid=2234, auid=501, euid=501, binary_path=/Applications/Bluetility.app/Contents/MacOS/Bluetility>}, requesting={<TCCDProcess: identifier=com.apple.bluetoothd, pid=740, auid=0, euid=0, binary_path=/usr/sbin/bluetoothd>}, },
default	09:30:04.438637-0800	bluetoothd	Bluetooth user permission alwaysAuth: granted
default	09:30:04.439464-0800	tccd	AUTHREQ_CTX: msgID=740.5, function=<private>, service=kTCCServiceBluetoothAlways, preflight=yes, query=1,
default	09:30:04.439502-0800	tccd	AUTHREQ_ATTRIBUTION: msgID=740.5, attribution={accessing={<TCCDProcess: identifier=com.rossible.Bluetility, pid=2234, auid=501, euid=501, binary_path=/Applications/Bluetility.app/Contents/MacOS/Bluetility>}, requesting={<TCCDProcess: identifier=com.apple.bluetoothd, pid=740, auid=0, euid=0, binary_path=/usr/sbin/bluetoothd>}, },
default	09:30:04.467495-0800	bluetoothd	BluetoothUserPrivacy: alwaysAccess: 1, isCBPrivacySupported: 1, medicalDeviceOrStateRestorationAppPendingTCC:0
default	09:30:04.467521-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.467538-0800	bluetoothd	Power state changed. PowerState:1 managerState:5 fUserAuthorization: 1	 medicalDeviceorStateRestorationAppPendingTCC:0
default	09:30:04.467562-0800	bluetoothd	Sending 'state updated' event with state "On" to session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.467577-0800	bluetoothd	Sending XPC message "CBMsgIdStateUpdated" to session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.467607-0800	bluetoothd	Sending XPC message "CBMsgIdSupportedFeatures" to session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.467642-0800	bluetoothd	advertisingAddressDidChange connectable:<private> non-connectable:<private>
default	09:30:04.467664-0800	bluetoothd	Ignore advertisingAddressDidChange for session <private> with access level : 0 and manager state not powered on
default	09:30:04.467740-0800	bluetoothd	advertisingAddressDidChange connectable:<private> non-connectable:<private>
default	09:30:04.467812-0800	bluetoothd	Ignore advertisingAddressDidChange for session <private> with access level : 0 and manager state not powered on
default	09:30:04.467909-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.468014-0800	bluetoothd	Received XPC message "CBMsgIdScan" from session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.468065-0800	bluetoothd	Setting client list to <private>
[... cut ...]
default	09:30:09.924926-0800	bluetoothd	allowDupes 0, agent: 0, active: 0
default	09:30:09.924940-0800	bluetoothd	active 0
default	09:30:09.924978-0800	bluetoothd	ScanParams:AD:0 AS:0 MSL:3 (30/300) PSV:1 [com.apple.bluetoothd-central-740-26 SO:0 AP:0 AD:0(30/300) AS:0 RAS:0 DMN:1 FG:0 ADVBF:0] |
default	09:30:09.924994-0800	bluetoothd	Returning scan parameters: Main:30.00ms/300.00ms LP:0.00ms/300.00ms Dm:(m:30/300) Cond:0:1:0 Sc:1:0 (passive) Dupfilter:Enabled minScanLevel:3 ENAssertion:N
default	09:30:09.925009-0800	bluetoothd	needToRestart=0
```

kdewald, 2022-01-23:
Ok, I have found a difference between the setup process of Bluetility and SimpleBLE that might give some indication of what is going on. Note the `fUserAuthorization:` value at the end of the process.

### BLUETILITY

```
default	09:30:04.220618-0800	bluetoothd	ReadyForTCC. TCC required : 1
default	09:30:04.220645-0800	bluetoothd	Sending XPC message "CBMsgIdReadyForTCC" to session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.220693-0800	bluetoothd	Session "com.rossible.Bluetility-central-2234-40" : needsRestrictedStateOperation = 0, overrideRestrictedState = 0 , denylistMode = 0, receivesControllerBTClockEvents=0
default	09:30:04.220735-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.376688-0800	Bluetility	Received CBMsgIdReadyForTCC
default	09:30:04.376738-0800	Bluetility	Running performTccCheck CBManager tccAvail 1, tccRequired 1
default	09:30:04.376761-0800	Bluetility	TCC required

default	09:30:04.377628-0800	tccd	AUTHREQ_CTX: msgID=2234.2, function=<private>, service=kTCCServiceBluetoothAlways, preflight=no, query=1,
default	09:30:04.408869-0800	bluetoothd	Received XPC message "CBMsgIdTCCDone" from session "com.rossible.Bluetility-central-2234-40"
default	09:30:04.410125-0800	tccd	AUTHREQ_CTX: msgID=740.4, function=<private>, service=kTCCServiceBluetoothAlways, preflight=yes, query=1,
default	09:30:04.410171-0800	tccd	AUTHREQ_ATTRIBUTION: msgID=740.4, attribution={accessing={<TCCDProcess: identifier=com.rossible.Bluetility, pid=2234, auid=501, euid=501, binary_path=/Applications/Bluetility.app/Contents/MacOS/Bluetility>}, requesting={<TCCDProcess: identifier=com.apple.bluetoothd, pid=740, auid=0, euid=0, binary_path=/usr/sbin/bluetoothd>}, },
default	09:30:04.438637-0800	bluetoothd	Bluetooth user permission alwaysAuth: granted
default	09:30:04.439464-0800	tccd	AUTHREQ_CTX: msgID=740.5, function=<private>, service=kTCCServiceBluetoothAlways, preflight=yes, query=1,
default	09:30:04.439502-0800	tccd	AUTHREQ_ATTRIBUTION: msgID=740.5, attribution={accessing={<TCCDProcess: identifier=com.rossible.Bluetility, pid=2234, auid=501, euid=501, binary_path=/Applications/Bluetility.app/Contents/MacOS/Bluetility>}, requesting={<TCCDProcess: identifier=com.apple.bluetoothd, pid=740, auid=0, euid=0, binary_path=/usr/sbin/bluetoothd>}, },
default	09:30:04.467495-0800	bluetoothd	BluetoothUserPrivacy: alwaysAccess: 1, isCBPrivacySupported: 1, medicalDeviceOrStateRestorationAppPendingTCC:0
default	09:30:04.467521-0800	bluetoothd	PowerManager power state is 1
default	09:30:04.467538-0800	bluetoothd	Power state changed. PowerState:1 managerState:5 fUserAuthorization: 1	 medicalDeviceorStateRestorationAppPendingTCC:0
```

### SIMPLEBLE

```
default	23:46:51.595541-0800	bluetoothd	ReadyForTCC. TCC required : 1
default	23:46:51.595601-0800	bluetoothd	Sending XPC message "CBMsgIdReadyForTCC" to session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.595647-0800	bluetoothd	Session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37" : needsRestrictedStateOperation = 0, overrideRestrictedState = 0 , denylistMode = 0, receivesControllerBTClockEvents=0
default	23:46:51.595676-0800	bluetoothd	PowerManager power state is 1
default	23:46:51.595979-0800	example_scan	Received CBMsgIdReadyForTCC
default	23:46:51.596016-0800	example_scan	Running performTccCheck CBManager tccAvail 1, tccRequired 1
default	23:46:51.596041-0800	example_scan	TCC required

default	23:46:51.597168-0800	tccd	AUTHREQ_CTX: msgID=2060.3, function=<private>, service=kTCCServiceBluetoothAlways, preflight=no, query=1,
default	23:46:51.628277-0800	example_scan	Application does have Bluetooth authorization.
default	23:46:51.631831-0800	bluetoothd	Received XPC message "CBMsgIdTCCDone" from session "example_scan-55554944272ccea7c927344db6a6d96f95df8646-central-2060-37"
default	23:46:51.631911-0800	bluetoothd	PowerManager power state is 1
default	23:46:51.631936-0800	bluetoothd	Power state changed. PowerState:1 managerState:5 fUserAuthorization: 0	 medicalDeviceorStateRestorationAppPendingTCC:0
```

nate2427, 2022-01-23:
@kdewald so what does this mean in terms of Bluetooth connectivity on macOS 12+? We should switch from the SimpleBLE API?

kdewald, 2022-01-23:
As far as I can tell, the problem is still related to the TCC (Transparency, Consent, and Control) subsystem within MacOS and the way it forwards permissions to the Bluetooth subsystem. This is something only Apple can fix, from what I can tell.

This problem seems to be happening to command line applications only, as anything packaged as an app bundle seems to be working fine (even the SimpleBLE examples, although they won't print anything when executed as a bundle).

kdewald, 2022-01-24:
For whoever finds this post:

I made a minimal example that shows this failure when running on MacOS Monterrey that you can find here: https://github.com/kdewald/MontereyBluetoothError

If you manage to make it work, please reach out so that I can bring those changes into SimpleBLE.

microbuilder, 2022-01-24:
@kdewald I'll see if I can ask a colleague who has a fair amount of experience with BLE on iOS/OSX to have a look at this, if he has a few hours available.

kdewald, 2022-01-25:
Ok, I'll start looking into how we can improve the library to add this functionality.

kdewald, 2022-01-30:
Everybody, great news! This issue seems to be fixed on Monterey 12.3! Thanks a lot for your help and patience!

I'll add a note and close this issue once and for all :P

Tagging @fidoriel @microbuilder for visibility.

closed 2022-01-30 by kdewald

microbuilder, 2022-03-14:
Just updated to 12.3 final and confirmed fixed.