simpleble/simpleble #52: Features for v0.4.0

kdewald, 2022-04-04:
- [ ] Proper behavior of uninitialized `Adapter` and `Peripheral` objects. Decide if either no-op or exception, add function to allow the user to check if initialized. Add tests.

johnboiles, 2022-04-15:
Let me know if you think I should put this in a separate issue, but I'd also love to see the ability to scan for specific services.

For macOS that means the plumbing to be able to get specific service IDs to https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/a2f3583f68161af34470c77c0ff7f28dfa47bdcc/src/macos/AdapterBaseMacOS.mm#L54

kdewald, 2022-04-18:
> Let me know if you think I should put this in a separate issue, but I'd also love to see the ability to scan for specific services.
> 
> For macOS that means the plumbing to be able to get specific service IDs to
> 
> https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/a2f3583f68161af34470c77c0ff7f28dfa47bdcc/src/macos/AdapterBaseMacOS.mm#L54
> 
> Alternately / additionally you could make available any advertised services with [`CBAdvertisementDataServiceDataKey`](https://developer.apple.com/documentation/corebluetooth/cbadvertisementdataservicedatakey?language=objc) from [`centralManager:didDiscoverPeripheral:advertisementData:RSSI:`](https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/a2f3583f68161af34470c77c0ff7f28dfa47bdcc/src/macos/AdapterBaseMacOS.mm#L96)

Hey @johnboiles, thanks for the request!

I'll tag this one as part of v0.5.0, as I do need to do some research as to how that functionality is exposed on Windows and Linux. One of the main goals of SimpleBLE is to maintain a uniform API whenever possible, so I need to be sure that I can get it to work in all cases with the same outwards-facing API.

Yohannfra, 2022-05-02:
Hi, 
Would it be possible to 'cheat' for features like that to ensure uniform API ? 

Like adding a function **scanForPeripheralsWithServices()** that internally does two things:
1. scan everything and get all scanned devices in a vector
2. filter out all the devices without the said service

kdewald, 2022-05-03:
Yes, what I was thinking about doing was adding the necessary parameters and ignoring them on the OSes that don't support this feature. We'll need to keep track of what is supported for each OS though :P

closed 2022-05-06 by kdewald

Symbitic, 2022-05-30:
If possible, I would simply like to see `services()` populated before calling `connect()`, or else have both a `scanForPeripheralsWithServices()` and `scanPeripheralsWithServices()`. I need the non-blocking API for language bindings.
Either way, one if not both should be available on all platforms. The cross-platform WebBluetooth standard requires filtering by services, so it must be available in some form on all platforms.

kdewald, 2022-05-30:
Hey @Symbitic, thanks for the request!

> I would simply like to see services() populated before calling connect()

Unfortunately, this specific thing cannot be achieved due to the fact that service resolution happens after a connection is established.

> or else have both a scanForPeripheralsWithServices() and scanPeripheralsWithServices()

Yes, scanning with services will be included in version 0.5.0. The only remaining thing for v0.4.0 is a weird race condition bug that I'm trying to track down.