simpleble/simpleble #60: Allow optionally building simpleble, simpleble-static, simple-c, simple-c-static

ElsterSolutionsOpensource, 2022-04-19:
It would be useful if the CMakeLists.txt offered options to control which artifacts are built: simpleble, simpleble-static, simple-c, simple-c-static. The reason for this is because on the Mac, we build our C++ framework as static libraries, which can be built without specifying a Developer Team. (If built as shared libraries, then the Developer Team has to be specified, and it it is not specified, the build fails). 

The options in CMakeLists.txt could look something like:

option (SIMPLEBLE_STATIC_EXCLUDE "Exclude simpleble-static" OFF)
option (SIMPLEBLE_EXCLUDE "Exclude simpleble" OFF)
option (SIMPLEBLE_C_STATIC_EXCLUDE "Exclude simpleble-c-static" OFF)
option (SIMPLEBLE_C_EXCLUDE "Exclude simpleble-c" OFF)

Then if's applied around each artifact. 

As an example, I've attached a CMakeLists.txt (from SimpleBLE release  v0.2.0) that I have already modified and tested to do this. 

[CMakeLists.txt](https://github.com/OpenBluetoothToolbox/SimpleBLE/files/8516176/CMakeLists.txt)

kdewald, 2022-04-19:
Yes! This is something that has already caused quite a bit of controversy, so I'm planning on going over this pretty soon.

kdewald, 2022-07-20:
Hey @ElsterSolutionsOpensource, wanted to quickly let you know that this has been added to the v0.5 preview,(currently on `feature/next` but soon in `main`. The selection will be based on using CMake's `BUILD_SHARED_LIBS` flag.

closed 2022-07-20 by kdewald

ElsterSolutionsOpensource, 2022-07-20:
Thanks @kdewald!