simpleble/simpleble #62: How to check if bluetooth is enabled

Yohannfra, 2022-04-28:
Hi,

On macos if I run 
```cpp
    auto adapter_list = SimpleBLE::Adapter::get_adapters();
```

with bluetooth enable or not I get the exact same result:
```
Default Adapter [00:00:00:00:00:00]
```

How can I detect if the bluetooth is enabled or not on the computer ? 

Thank you

kdewald, 2022-04-29:
This specific problem is a bit tricky, because the underlying API is soooo different for each operating system. There is an MR that has [solved this problem for MacOS](https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/34), but I haven't had time to look into equivalents for Windows and Linux.

The fact that you always get one adapter regardless of the BLE state is a horrible workaround I made, which I apologize for. I will dig deeper into this once I get v0.4.0, which is almost there besides from a very specific memory leak that I just can't find.

kdewald, 2022-09-15:
This has been fixed with the provided `Adapter::bluetooth_enabled()` function.

closed 2022-09-15 by kdewald