simpleble/simpleble #65: (MacOS) WriteRequest failing to send message.

kdewald, 2022-04-29:
Not 100% confirmed, but it seems like calling WriteRequest (which internally is calling `[self.peripheral writeValue:payload forCharacteristic:characteristic type:CBCharacteristicWriteWithoutResponse];` seems to be not generating a BLE write packet under certain (unknown) circumstances. This has been validated with the packet logger.

kdewald, 2022-04-29:
Ok, it seems the current implementation is incorrect. Need to see the following documents:
- https://developer.apple.com/documentation/corebluetooth/cbperipheral/2891512-cansendwritewithoutresponse
- https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/2874034-peripheralisready

g8kig, 2022-04-30:
Fix confirmed with latest change 9585001de9b525cdca80e1dabe7d494d1fff7fb7 thank you!

kdewald, 2022-05-02:
Fixed on: https://github.com/OpenBluetoothToolbox/SimpleBLE/commit/e4e63d156ff2d040e42b347100c8c06de7512545

closed 2022-05-02 by kdewald