simpleble/simpleble #66: Request: Add a non-blocking API.

Symbitic, 2022-05-07:
This is a request for a feature I'd like to see.
It would really help for creating bindings to Javascript/Kotlin if there was a way to avoid blocking the main thread from the caller.
Something like:
```
adapter.start_scanning_for(5000);
while (!adapter.is_done()) {
    adapter.update_scan();
}
std::vector<SimpleBLE::Safe::Peripheral> peripherals = adapter.get_devices()
    .value_or(std::vector<SimpleBLE::Safe::Peripheral>());
```

I was working on Deno FFI bindings for SimpleBLE. I had to abandon it for the time because it freezes the entire program until `adapter.scan_for` has finished.

kdewald, 2022-05-08:
Hey @Symbitic, thanks for reaching out!

There is a non-blocking alternative to `scan`, which is to use `scan_start` and `scan_stop`. They are the non-blocking version of this API and it will be up to you to handle the language-specific non-blocking delay in the middle part, as every language is different in this regard.

Let me.know if you need help, I want to see your project succeed!

kdewald, 2022-05-17:
Closing due to lack of activity. @Symbitic feel free to reach out if you need help.

closed 2022-05-17 by kdewald