simpleble/simpleble #67: Ubuntu sigabort during write_request

Andrey1994, 2022-05-11:
Hi, could you take a brief look at https://github.com/brainflow-dev/brainflow/issues/434 ?

I am not 100% sure that its an issue inside SimpleBLE but I guess so, and it works on other OSes. I dont fully understand the backtrace, it seems like one thread sleeps and another one crashes somewhere...

```
#4  0x00007ffff7b3c3f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff767fee8 in SimpleBLE::Safe::Peripheral::write_request (this=0x7ffff00174b0, service="0000fe8d-0000-1000-8000-00805f9b34fb", characteristic="273e0001-4c4d-454d-96be-f03bac821358", data="\003v1\n")
    at /home/username/projects/realtime_EEG/brainflow/third_party/SimpleBLE/src/PeripheralSafe.cpp:86
```
I supposed there should be some methods called between these two lines

Also, its not the latest SimpleBLE fork, I will update it soon but looking at commits in upstream it seems like there were no relevant changes and fixes

kdewald, 2022-05-14:
hey @Andrey1994 , thanks for the report!

I just checked, Brainflow is using a pre-release of SimpleBLE v0.1.1, which is using the old SimpleBluez/SimpleDBus backend, which was completely rewritten and released in SimpleBLE v0.3.0, precisely due to all of these instability issues. It's very hard to know what could be wrong with that version, so I'd heavily encourage to upgrade to the latest version on the `main` branch. (I'm still iterating on a few edge cases, but it should be pretty stable at this point).

Let me know if things are still broken after the upgrade!

Andrey1994, 2022-05-16:
Thanks a lot, the issue is solved

closed 2022-05-16 by Andrey1994