simpleble/simpleble #73: Feature request: descriptors

Symbitic, 2022-06-22:
I've been working on bindings for the JavaScript runtime Deno. You can view my work here: https://github.com/Symbitic/deno-webbluetooth

First off, *thank you so, so much for the cross-platform BLE library. This wouldn't have been possible without you.*

It isn't fully compliant with the WebBluetooth spec yet. The one big thing I am missing right now is descriptors. Is it possible to add cross-platform support for them?

More information about WebBluetooth:
https://developer.mozilla.org/en-US/docs/Web/API/Web_Bluetooth_API
https://webbluetoothcg.github.io/web-bluetooth

kdewald, 2022-06-23:
Hey @Symbitic, thanks a lot for the feedback! I really appreciate it!

I think it's possible to extract the descriptor information in all 3 operating systems, but I do need to do some research on how that data is presented. I've added it to my current task tracker here: https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/59

kdewald, 2022-06-24:
Also @Symbitic, I've been wanting to provide a JS wrapper for SimpleBLE for a while. If you could guide me a bit on how that is set up, I'd appreciate it a lot!

Symbitic, 2022-06-29:
Hey @kdewald.

The JS wrapper isn't universal: this one only works for Deno. However, it should be relatively simple to make Node.js bindings as well, either with FFI or with [NAPI](https://www.npmjs.com/package/node-addon-api).

Descriptors should be possible. QtBluetooth has them: https://code.qt.io/cgit/qt/qtconnectivity.git/tree/src/bluetooth

To give you a bit of an overview: Deno has an (unstable) FFI API for making native calls. The first part is to build the simpleble-c shared library. Locally, it is built as a submodule. But when loaded remotely (e.g. `import { WebBluetooth } from "https://deno.land/x/webbluetooth@v0.1.2/mod.ts"`) it will download the shared library (saving the need to build it on every machine). This is why release versions upload the shared library to GitHub releases: https://github.com/Symbitic/deno-webbluetooth/releases/tag/v0.1.2

The FFI symbols are declared here: https://github.com/Symbitic/deno-webbluetooth/blob/main/src/symbols.ts
They are loaded and wrapped into functions here: https://github.com/Symbitic/deno-webbluetooth/blob/main/src/ffi.ts

Where it gets difficult is with pointers and data structures. Numbers (returns and params) are pretty simple, but pointers are either a bigint (refers to address in memory) or a Uint8Array. Opaque pointers (like `simpleble_peripheral_t`) are bigints, while pointers like strings are encoded/decoded into a Uint8Array. For 16-bit (and higher) values, `DataView` allows setting/getting 16-bit numbers in a Uint8Array, and it even handles endian conversion when needed.

For data structures, a Uint8Array of the same size as the data structure is created, then values are get/set correspondingly (`const manufacturerDataLen = view.getBigUint64(8, true)`; 8 = offset in `simpleble_manufacturer_data_t.data_length`)

The functions that convert normal JS/TS values and objects into FFI arrays are in `ffi.ts`; the rest of the WebBluetooth wrapper uses them.

This isn't completely compliant with the WebBluetooth standard yet.
Deno doesn't support FFI callbacks yet, so some of the functions can't be used. I'm really looking forward to when that gets added to Deno.

If you've installed the latest version of Deno, then try this:

    deno run -A --unstable https://denopkg.com/symbitic/deno-webbluetooth@v0.1.3/examples/simpleble-scan.ts

kdewald, 2022-07-04:
Thanks! I'll definitely look into this in more depth.

I have another question: What features of descriptors would you need for this? Could you write some stub of the expected API? I can use that as a starting point to investigate how I can expose that in a common fashion across all operating systems.

Symbitic, 2022-07-04:
Hey. I just need to get a list of descriptors for a characteristic and read/write values to descriptors.

Something like this should do:
```c
size_t simpleble_peripheral_descriptor_count(simpleble_peripheral_t handle, 
                                             simpleble_uuid_t service,
                                             simpleble_uuid_t characteristic);
simpleble_err_t simpleble_peripheral_descriptor_get(simpleble_peripheral_t handle, 
                                                    simpleble_uuid_t service,
                                                    simpleble_uuid_t characteristic,
                                                    size_t index,
                                                    simpleble_uuid_t* descriptor);
simpleble_err_t simpleble_peripheral_descriptor_read(simpleble_peripheral_t handle, 
                                                     simpleble_uuid_t service,
                                                     simpleble_uuid_t characteristic,
                                                     simpleble_uuid_t descriptor,
                                                     uint8_t** data, size_t* data_length);
simpleble_err_t simpleble_peripheral_descriptor_read(simpleble_peripheral_t handle, 
                                                     simpleble_uuid_t service,
                                                     simpleble_uuid_t characteristic,
                                                     simpleble_uuid_t descriptor,
                                                     uint8_t* data, size_t data_length);
```

It should be possible to do this on all platforms.
Windows has `GattCharacteristic.GetDescriptorsAsync` and `GattDescriptor`.
macOS has `CBPeripheral discoverDescriptorsForCharacteristic/readValueForDescriptor/writeValue`.
Linux BlueZ has `org.bluez.GattDescriptor1` for reading/writing, and a list of descriptors can be obtained from `/org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX/serviceXXXX/charYYYY/descZZZZ`.

I'm not sure if callbacks for when a descriptor/characteristic value has changed are possible on all platforms, but that's optional.

Some random links that may help:
+ https://github.com/abandonware/noble/blob/master/lib/win/src/ble_manager.cc
+ https://github.com/abandonware/noble/blob/master/lib/mac/src/ble_manager.mm
+ https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt
+ https://code.qt.io/cgit/qt/qtconnectivity.git/tree/src/bluetooth/bluez/org.bluez.GattDescriptor1.xml

Deno finally has callbacks, so I've been adding more bindings.

Symbitic, 2022-07-09:
@kdewald I made a pull request for SimpleBluez that should add everything needed. I don't have access to windows or mac ATM so I can't do anything for them.

I'll try to work on porting this to Node.js and the newer [bun.sh](https://bun.sh/) as well when I have this ready.

Thanks again for this library. One of the things I'm planning on using it for is a library for the LEGO Powered Up devices that use BLE.

kdewald, 2022-07-21:
Hey @Symbitic, thanks for all the contributions!

I'm cleaning up a few things that I need to make the addition of descriptors clean and organized, which might require some breaking API changes if we want to be consistent with function naming. I hope to have a first pass soon, I'll keep you posted!

kdewald, 2022-07-23:
Hey @Symbitic, you can track my progress for descriptors here: https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/81

Symbitic, 2022-07-23:
Hey @kdewald I just submitted a pull request to add descriptors to all platforms: #82

I didn't realize you were working on it until after I submitted it.
It doesn't have any breaking changes, and it keeps the naming consistent.

Hope you find it useful.

kdewald, 2022-07-27:
As I mentioned in the MR, your code was super useful to speed this process up. Everything is landed, so it's time to start testing :P

closed 2022-07-27 by kdewald