simpleble/simpleble #86: simpleble_peripheral_write_descriptor is not defined on Linux

Symbitic, 2022-07-30:
When I build SimpleBLE-C as a shared library (`-DBUILD_SHARED_LIBS=ON`), the `simpleble_peripheral_write_descriptor` function is not defined in the shared library.

Example:
```
$ nm -D ./build/bin/libsimpleble-c.so | grep -i simpleble_peripheral_
0000000000007bc0 T simpleble_peripheral_read
0000000000009490 T simpleble_peripheral_read_descriptor
00000000000083f0 T simpleble_peripheral_write_command
0000000000007fc0 T simpleble_peripheral_write_request
0000000000009a60 T _Z37simpleble_peripheral_write_descriptorPv16simpleble_uuid_tS0_S0_PKhm
```

For some reason, I get similar on the `exp/descriptors` branch:
```
$ nm -D ./build/bin/libsimpleble-c.so | grep -i simpleble_peripheral_
0000000000007780 T simpleble_peripheral_address
00000000000078b0 T simpleble_peripheral_connect
0000000000007d40 T simpleble_peripheral_descriptor_count
0000000000008100 T simpleble_peripheral_descriptor_get
0000000000008ee0 T simpleble_peripheral_write_command
0000000000008ab0 T simpleble_peripheral_write_request
0000000000009f80 T _Z31simpleble_peripheral_read_valuePv16simpleble_uuid_tS0_S0_PPhPm
000000000000a550 T _Z32simpleble_peripheral_write_valuePv16simpleble_uuid_tS0_S0_PKhm
```

Not sure what the cause is. It *might* be fixable using [GenerateExportHeader](https://cmake.org/cmake/help/latest/module/GenerateExportHeader.html).

kdewald, 2022-07-30:
I think there's a typo in one of the functions. Looking into it now.

kdewald, 2022-07-30:
Ok, it seems that the functions are being name-mangled. Not sure why.

Symbitic, 2022-07-30:
Thanks. BTW, the `SimpleBLE::Characteristic& characteristic` on line 156 in peripheral.cpp is giving me weird problems.

This always causes a segfault:
```c++
SimpleBLE::Characteristic& characteristic = service.characteristics()[i];
memcpy(services->characteristics[i].uuid.value, characteristic.uuid().c_str(), SIMPLEBLE_UUID_STR_LEN);
```
But this works perfectly for some reason:
```c++
memcpy(services->characteristics[i].uuid.value, service.characteristics()[i].uuid().c_str(), SIMPLEBLE_UUID_STR_LEN);
```

Here's the full loop I have:
```c++
    for (size_t i = 0; i < services->characteristic_count; i++) {
        memcpy(services->characteristics[i].uuid.value, service.characteristics()[i].uuid().c_str(), SIMPLEBLE_UUID_STR_LEN);

        services->characteristics[i].descriptor_count = service.characteristics()[i].descriptors().size();

        if (services->characteristics[i].descriptor_count > SIMPLEBLE_DESCRIPTOR_MAX_COUNT) {
            services->characteristics[i].descriptor_count = SIMPLEBLE_DESCRIPTOR_MAX_COUNT;
        }

        for (size_t j = 0; j < services->characteristics[i].descriptor_count; j++) {
            memcpy(services->characteristics[i].descriptors[j].uuid.value, service.characteristics()[i].descriptors()[j].uuid().c_str(),
                   SIMPLEBLE_UUID_STR_LEN);
        }
    }
```

kdewald, 2022-07-31:
Ok, I can reproduce the problem locally. Looking into it.

kdewald, 2022-07-31:
@Symbitic I just fixed the segfault issue. Do you still have problems consuming the library as a shared one?

Symbitic, 2022-08-02:
@kdewald It turned out to be so, so simple. The declaration of `simpleble_peripheral_write_descriptor` in peripheral.h uses `uint8_t* data`, while the implementation in peripheral.cpp uses `const uint8_t* data`.

Just change `uint8_t* data` to `const uint8_t* data` in peripheral.h, and it should be fine.

kdewald, 2022-08-02:
Damn, that's on me. I just uploaded the fix to `main`.

closed 2022-08-02 by kdewald