simpleble/simpleble #88: CMake build is broken

Symbitic, 2022-08-15:
Building is broken for me. Whenever I run it, it gives me this:
```bash
$ cmake --build build --verbose
[ 51%] Building CXX object CMakeFiles/simpleble.dir/src/Adapter.cpp.o
/usr/bin/c++ -DFMT_HEADER_ONLY=1 -I/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/include -I/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/src -I/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/src/builders -I/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/src/external -I/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/src/linux -I/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/build/_deps/fmt-src/include -I/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/build/_deps/simplebluez-src/include -fPIC -std=gnu++17 -MD -MT CMakeFiles/simpleble.dir/src/Adapter.cpp.o -MF CMakeFiles/simpleble.dir/src/Adapter.cpp.o.d -o CMakeFiles/simpleble.dir/src/Adapter.cpp.o -c /home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/src/Adapter.cpp
In file included from /home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/src/linux/AdapterBase.h:9,
                 from /home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/src/Adapter.cpp:3:
/home/alex/Downloads/Software/ble/deno-webbluetooth/SimpleBLE2/build/_deps/simplebluez-src/include/simplebluez/Adapter.h:3:10: fatal error: simpledbus/advanced/Proxy.h: No such file or directory
    3 | #include <simpledbus/advanced/Proxy.h>
```
The problem seems to be the SimpleBluez CMake target not exporting the include directories. SimpleBluez builds fine, but not SimpleBLE.

kdewald, 2022-08-15:
Yes, that was me. I'll revert the broken changes tonight.

kdewald, 2022-08-16:
Ok, everything should be reverted by now. My bad for not testing this feature end to end before landing.

Symbitic, 2022-08-17:
Thanks. BTW, for the CI tests that are failing: you don't need to use mkdir and cd.
Just use:
```bash
cmake -B build64 -S sourcedir
cmake --build build64
```
CMake will handle creating the dir.
See how I do it here: https://github.com/Symbitic/deno-webbluetooth/blob/main/.github/workflows/test.yml

kdewald, 2022-08-18:
Great piece of info, thanks!

I'm very close to having a pretty big reorg of the code that will need this to be fixed as well. Bear with me for a little longer, great things are coming our way :D

Symbitic, 2022-08-22:
@kdewald I'll bear with you for as long as you want. Your project, I just use your hard work, so thank you.

I notice that the new/layout branch uses symbol hiding. If you use that, you need to mark each public class/function as exported, or the symbol will be hidden in the shared library. CMake has a module to help: https://cmake.org/cmake/help/latest/module/GenerateExportHeader.html
I'm working on a pull request for it. I already submitted another filtering scans based on service UUIDs.

kdewald, 2022-08-23:
> @kdewald I'll bear with you for as long as you want. Your project, I just use your hard work, so thank you.

Thanks for all the support!! This is what makes this project worthwhile, the fact that I can enable others in their work :D

> I notice that the new/layout branch uses symbol hiding. If you use that, you need to mark each public class/function as exported, or the symbol will be hidden in the shared library. CMake has a [...]

That's correct, but I've only enabled hiding of inline functions, which should not have any negative effects. I hope at some point to add proper export headers too, although there are a few minor considerations to take into account based on the fact that SimpleBLE is importing source files from SimpleBluez and SimpleDBus too, where it might be interesting to toggle visibility if necessary.

This being said, all contributions are welcome! Incoming code cuts down my research and implementation times quite a lot, so we can cover more terrain faster.

Symbitic, 2022-08-30:
@kdewald it's building again, but please try to add the `NO_SONAME` and `BUILD_SIMPLEBLE_SHARED`/`BUILD_SIMPLEBLE_C_SHARED` options from my PR. I need SimpleBLE-C to be the only one built as a shared library, so it includes all the other dependencies.

kdewald, 2022-09-02:
Can you provide a bit more detail on how you're using this? I think I'm missing a few details of what is necessary.

Symbitic, 2022-09-02:
Like I said, I no longer need any of those things.

closed 2022-09-03 by kdewald