simpleble/simpleble #90: Expose GATT Characteristic Properties in Characteristic class

jferdelyi, 2022-08-18:
I think it would be a good idea to have the GATT characteristic properties directly accessible in the Characteristic class.

kdewald, 2022-08-18:
Hey @jferdelyi, thanks for the feedback!

Could you elaborate a bit more on what the desired API would look like? I'm not entirely sure I understood what you expect.

Thanks!

jferdelyi, 2022-08-18:
Maybe I miss something, my idea is relatively simple: for each caracteristic be able to get all properties like Write, Notify, Read, etc.

In the actual implementation, as far as I know, the only way to know this is to try, and an exception is raised if the caracteristic doesn't have the propertie (or nullopt in the case of "safe" usage).

kdewald, 2022-08-18:
Oh, I see! Yeah, that's a very reasonable request. Added to my list :)

ElsterSolutionsOpensource, 2022-09-05:
I would also love to have the library report the characteristic properties as my tool works with multiple peripherals. Currently, they all use write without response and notify to exchange data. I'd like my tool to be able to verify which characteristic properties the peripheral supports, and use the appropriate services or throw an error, as appropriate. 
Thanks Kevin for this great library!

kdewald, 2022-11-14:
Implemented in https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/130

closed 2022-11-14 by kdewald