simpleble/simpleble #91: Add the list of company identifiers

jferdelyi, 2022-08-18:
This is just a suggestion, I'm not sure if this is a good idea to incorporate into the project.

Maybe adding the list of company identifiers might be a good idea to convert the manufacturer data code into the company name string. The official list can be found here (in CSV):
https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/

The "problem" is that this list will continue to grow and that a regular update is necessary...

kdewald, 2022-08-19:
Good suggestion. Keeping the library updated shouldn't be hard. I'll add it to the list.

Yohannfra, 2022-09-14:
Hi,

I needed the same feature for my project and implemented a solutions a few months ago that could be use for this library.

Here : https://github.com/Yohannfra/generate_bluetooth_16_bits_uuid_json is a python script that parses the bluetooth sig pdf of all the uuid and generate a json that I update frequently (you can find it [here](https://github.com/Yohannfra/generate_bluetooth_16_bits_uuid_json/blob/master/ble_16_bits_uuids.json)

Once the json is generated it's trivial to parse it and use it as a map to find the names.

Yohannfra, 2022-09-14:
It has the company names but also all the 16-bit UUID of the standard services and characteristics

Yohannfra, 2022-09-14:
My bad it doesn't have the company identifiers but only the members, I will update it soon to also add the company identifiers from the csv

kdewald, 2022-09-15:
@Yohannfra FYI, I found this: https://github.com/NordicSemiconductor/bluetooth-numbers-database

Yohannfra, 2022-10-26:
> @Yohannfra FYI, I found this: https://github.com/NordicSemiconductor/bluetooth-numbers-database

Thanks for the link, this is much better than what I have I'll use this instead.

kdewald, 2026-10-27:
Closing this here and tracking it on our internal board, following the plan in the comment above. It's still on the list.

closed 2026-10-27 by kdewald