simpleble/simpleble #94: SimplePyBLE and Windows attribute caching

spkurtzweil, 2022-09-03:
It seems calling peripheral.read() on a characteristic reads old cached data; reconnecting the device refreshes the data. Would it be possible to add an option to read uncached?

kdewald, 2022-09-03:
Hey @spkurtzweil, thanks for the report!

This issue has already been fixed in SimpleBLE, but I haven't been able to make a new release for SimplePyBLE yet (I'm almost there, I swear). I'll provide a link to the fixed version once it's out.

spkurtzweil, 2022-09-03:
Understood. Thanks for the quick response!

kdewald, 2022-09-04:
Hey @spkurtzweil, just a quick update on this:

I put up a development version with the latest code on `main` that should have this issue fixed. You can find the installation instructions here: https://pypi.org/project/simplepyble/0.0.6.dev3/

Let me know how it goes!

closed 2022-09-04 by kdewald

spkurtzweil, 2022-09-04:
The value does update correctly now. Thank you.