simpleble/simpleble #110: Buffer overflow in simpleble_manufacturer_data_t

DrSegatron, 2022-09-21:
in [include/simpleble_c/types.h](https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/main/simpleble/include/simpleble_c/types.h):

```
typedef struct {
    uint16_t manufacturer_id;
    size_t data_length;
    uint8_t data[24];
} simpleble_manufacturer_data_t;
```
data can be overflowed as the data excluding ID and length can be 28 bytes.

kdewald, 2022-09-23:
Hey @DrSegatron, thanks for the report!

I ran the numbers on my end, I got 26 as the maximum length of data (aside from the manufacturer ID) that can be provided. This was my math:

```
    // Note: The maximum length of a BLE advertisement is 31 bytes.
    // The first byte will be the length of the field,
    // the second byte will be the type of the field (0xFF for manufacturer data),
    // the next two bytes will be the manufacturer ID,
    // and the remaining 26 bytes are the manufacturer data.
```

I've updated the number with these calculations, but if you see a mistake in my reasoning please let me know.

kdewald, 2022-09-23:
This is the commit with the change: https://github.com/OpenBluetoothToolbox/SimpleBLE/commit/bed98cde124d275b23645b7a48baecf3c0fcd200

DrSegatron, 2022-09-23:
Yes that checks out, I forgot the ff byte.

kdewald, 2022-09-23:
Awesome, thanks again for the report!

closed 2022-09-23 by kdewald