simpleble/simpleble #111: Add support for GAP Scan Response

DrSegatron, 2022-09-21:
I am missing support to get Scan Response data from GAP, in a manner similar to `simpleble_peripheral_manufacturer_data_get()`.

kdewald, 2022-09-23:
Hey @DrSegatron, thanks for the report.

Is the feature you're requesting similar to this one? https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/100
If not, could you provide some more details?

Thanks!

kdewald, 2022-09-24:
Awesome, thanks for the info! This is definitely worth adding.

Do you know how this feature is exposed by Bluez / WinRT / CoreBluetooth? Any pointers will help me implement this quicker.

DrSegatron, 2022-09-26:
Don't know anything about those, but it is usually implemented as a scan setting, passive or active. When active and it receives an advertisement, a SCAN is automatically sent which results in a SCAN RESPONSE and the data is accessible the same way the normal GAP advertisement data is.

kdewald, 2022-09-26:
It's very likely that this is handled transparently by the operating system and the data is already forwarded to SimpleBLE in advertising events, although I'm certain that I'm not handling this appropriately in all cases. It would actually explain why on Windows certain devices don't show their name even though it's correctly advertised.

I'll investigate more and share my notes.

kdewald, 2022-11-14:
Implemented in https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/130

closed 2022-11-14 by kdewald