simpleble/simpleble #112: Inconsistent freeing in the c examples

DrSegatron, 2022-09-21:
For example, in [examples/simpleble/c/scan/scan.c](https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/main/examples/simpleble/c/scan/scan.c), `adapter_on_scan_found()` and `adapter_on_scan_updated()` allocate the same strings, but does not free the same things:

```
    // Let's not forget to release the associated handles and memory
    simpleble_peripheral_release_handle(peripheral);
    simpleble_free(peripheral_address);
    simpleble_free(peripheral_identifier);
```

```
    // Let's not forget to release the associated handle.
    // Let's not forget to release the associated handles and memory
    simpleble_free(peripheral_address);
    simpleble_free(peripheral_identifier);
```

kdewald, 2022-09-23:
Hey @DrSegatron, thanks for the report! I just landed the fix.

DrSegatron, 2022-09-27:
Are you sure that 

`char* adapter_identifier = simpleble_adapter_identifier(adapter);`

shoudn't be freed in

`adapter_on_scan_found()` and `adapter_on_scan_updated()`?

And should the handle really be freed in `adapter_on_scan_found()`?

`simpleble_peripheral_release_handle(peripheral);`

kdewald, 2022-11-16:
@DrSegatron I looked at the examples and they seemed correct. Can you please point out which part you think is wrong? I'll close the issue in the meantime, but feel free to reopen.

closed 2022-11-16 by kdewald

DrSegatron, 2022-11-16:
Perhaps you should add comments to the free()s what you are freeing and why.

I might have mixed up this and the connect example, so the handles should be freed here as they are not needed after they are printed.

And is simpleble_adapter_identifier() only returning a pointer to a constant string and doesn't need to be freed?