simpleble/simpleble #114: Add Android support

kdewald, 2022-09-27:


ronaaron, 2023-08-09:
Very much would like to see this...

joemarshall, 2024-02-28:
For this to happen, it would need some JNI code to call through to android BLE support, which all lives in Java / Kotlin land. The simplest way I can see to do this is:

1) Build an android .aar or .jar which implements the bits of the BLE API needed here with JNI friendly callbacks. The callbacks in android BLE are all to classes derived from abstract base classes. Which you can't implement purely in C++ through java native because the methods aren't defined as native. So there needs to be some java code in the project (and the associated gradle build rubbish unfortunately...)

2) Make a lightweight JNI wrapper in C++ for the aar to call through to this, which loads the android library and passes everything off to it (and receives callbacks through native implementations of callback classes). 

I can see how this would be done, if I had more time I'd work on it, but I'm snowed under with paid work right now, and until someone asks me to port something with BLE to iOS, I will probably have to deal with my hacked together android specific BLE library for now.

kdewald, 2024-03-03:
Hey all, thanks for reaching out. This is definitely becoming one of the first priorities once we get the new licensing stuff worked out. We're aiming to have a lot more resources going into this in the near future.