simpleble/simpleble #132: SimpleBLE failed to reconnect after disconnection on Windows

kdewald, 2022-11-14:
Description of the problem: `I can scan for and find my device, and then I can connect to it and it works as intended, but if I disconnect and try to connect to the same device it returns this error`

[image]

klaff, 2022-11-28:
As a workaround, if you delete the peripheral (I'm working in python so that looks like `del(p)`) the connection will be closed, and you can then create a new peripheral object and use it to connect to the same Bluetooth device.

felixdollack, 2022-12-23:
I experience the same using C bindings on Windows. After closing and restarting my application connecting works again (once).

MacOS works without fail every time.

lairdrt, 2023-01-04:
Using Windows 10 SimpleBLE, exact same problem. The connection **is** being closed (the disconnect appears to work), that's not the problem at least for me, If you call peripheral.disconnect(), then something down deep is being released, and I cannot re-connect at all. The exception message is "The object has been closed." My code is as follows, from some sample:
```
#include "pch.h"

#include <iostream>
#include <vector>

#include <simpleble\SimpleBLE.h>

using namespace winrt;
using namespace Windows::Foundation;

int __stdcall wWinMain(HINSTANCE, HINSTANCE, LPWSTR, int)
{
    init_apartment(apartment_type::single_threaded);

    // Find Bluetooth adapter
    if (!SimpleBLE::Adapter::bluetooth_enabled()) {
        OutputDebugStringA("Bluetooth is not enabled\n");
    }
    else {
        OutputDebugStringA("Bluetooth is enabled\n");
    }
    auto adapters = SimpleBLE::Adapter::get_adapters();
    if (adapters.empty()) {
        OutputDebugStringA("No Bluetooth adapters found\n");
    }
    else {
        OutputDebugStringA("Bluetooth adapters found\n");
        for (auto a = adapters.begin(); a != adapters.end(); ++a) {
            OutputDebugStringA("Adapter:");
            OutputDebugStringA(a->identifier().c_str());
            OutputDebugStringA("\n");
            OutputDebugStringA("Address:");
            OutputDebugStringA(a->address().c_str());
            OutputDebugStringA("\n");
        }

        // Find Bluetooth peripherals
        SimpleBLE::Adapter adapter = adapters[0];

        // Scan for peripherals for 5000 milliseconds
        adapter.scan_for(5000);

        // Get the list of peripherals found
        std::vector<SimpleBLE::Peripheral> peripherals = adapter.scan_get_results();

        // Print the identifier of each peripheral
        for (auto p : peripherals) {
            OutputDebugStringA("Peripheral:");
            OutputDebugStringA(p.identifier().c_str());
            OutputDebugStringA("\n");
            OutputDebugStringA("Address:");
            OutputDebugStringA(p.address().c_str());
            OutputDebugStringA("\n");
        }

        // Scope block
        {
            SimpleBLE::Peripheral p1 = peripherals.front();

            p1.connect();
            OutputDebugStringA("Connected (1)\n");

            std::vector<std::pair<SimpleBLE::BluetoothUUID, SimpleBLE::BluetoothUUID>> uuids;
            for (auto service : p1.services()) {
                for (auto characteristic : service.characteristics()) {
                    uuids.push_back(std::make_pair(service.uuid(), characteristic.uuid()));
                }
            }

            OutputDebugStringA("The following services and characteristics were found:\n");
            for (size_t i = 0; i < uuids.size(); i++) {
                OutputDebugStringA(uuids[i].first.c_str());
                OutputDebugStringA(" / ");
                OutputDebugStringA(uuids[i].second.c_str());
                OutputDebugStringA("\n");
            }

            p1.disconnect();
            OutputDebugStringA("Disconnected (1)\n");
        }

        // New scope block
        {
            SimpleBLE::Peripheral p2 = peripherals.front();

            p2.connect(); // EXCEPTION: The object has been closed.
            OutputDebugStringA("Connected (2)\n");
        }
    }

    Uri uri(L"http://aka.ms/cppwinrt");
    ::MessageBoxW(::GetDesktopWindow(), uri.AbsoluteUri().c_str(), L"C++/WinRT Desktop Application", MB_OK);
}
```

lairdrt, 2023-01-04:
The Close() statement within file  SimpleBLE/simpleble/src/backends/windows/PeripheralBase.cpp and function PeripheralBase::disconnect() is causing (directly or indirectly) the problem; commenting out the Close() allows one to connect/disconnect/connect peripherals as would be expected. I would suspect the solution **is not** to delete the Close() statement, but it does seem to be causing the issue.
```
void PeripheralBase::disconnect() {
    gatt_map_.clear();
    if (device_ != nullptr) {
        device_.Close();
    }

    std::unique_lock<std::mutex> lock(disconnection_mutex_);
    if (disconnection_cv_.wait_for(lock, 5s, [=] { return !this->is_connected(); })) {
        // Disconnection successful
    } else {
        // Failed to disconnect
        // TODO: Raise an error
    }
}
```

felixdollack, 2023-01-05:
I just found a related issue in a different project. They mention a bug in the windows bluetooth API:
https://github.com/jasongin/noble-uwp/issues/20#issuecomment-313590205

The chromium project also created a workaround picking up the conversation mentioned above:
https://github.com/chromium/chromium/commit/9ab40ee7f3e6dfcdd1c0b597f8ef6a13bd42117e

@kdewald Is it possible to add some workarounds similar to what is discussed in those projects?

kdewald, 2023-01-20:
I am able to reproduce this systematically. Should be able to come up with a fix soon.

Gnome16, 2023-01-21:
If you re-run the scan for devices then you should find it reconnects again OK. I had a similar issue on Windows with some other BLE software - I couldn't connect to a device without doing a scan even if I knew its address.

 It might be related to how Windows caches the Bluetooth device data in which case there might not be a fix.

felixdollack, 2023-01-21:
Re-scanning does not work for me.
The previously found device seem to be cached, because I get the exact same list of found devices almost instantly.

Gnome16, 2023-01-21:
In the Pascal SimpleBleConnect test program, this works for me:

```
procedure TSimpleBleConnectExample.DoRun;
var
  ErrorMsg: String;
  Adapter: TSimpleBleAdapter;
  ErrCode: TSimpleBleErr = SIMPLEBLE_SUCCESS;
  i, j, k, Selection, ServicesCount: Integer;
  Peripheral: TSimpleBlePeripheral;
  PeripheralIdentifier: PChar;
  PeripheralAddress: PChar;
  Service: TSimpleBleService;
  Cnt : integer;
begin
  // quick check parameters
  ErrorMsg:=CheckOptions('h', 'help');
  if ErrorMsg<>'' then begin
    ShowException(Exception.Create(ErrorMsg));
    Terminate;
    Exit;
  end;

  // parse parameters
  if HasOption('h', 'help') then begin
    WriteHelp;
    Terminate;
    Exit;
  end;

  // look for BLE adapters
  if simpleble_adapter_get_count() = 0 then
  begin
    WriteLn('No BLE adapter was found.');
    Terminate;
    Exit;
  end;

  for cnt:=1 to 4 do begin

     PeripheralListLen:=0;

     // get a handle for the BLE Adapter
     Adapter := simpleble_adapter_get_handle(0);
     if Adapter = 0 then
     begin
       WriteLn('Could not get handle for BLE adapter.');
       Terminate;
       Exit
     end;
     WriteLn('Found BLE adapter and got handle.');

     // register SimpleBLE scan callback functions
     simpleble_adapter_set_callback_on_scan_start(Adapter, @AdapterOnScanStart, Nil);
     simpleble_adapter_set_callback_on_scan_stop(Adapter, @AdapterOnScanStop, Nil);
     simpleble_adapter_set_callback_on_scan_found(Adapter, @AdapterOnScanFound, Nil);

     // start BLE scanning for 5 seconds
     simpleble_adapter_scan_for(Adapter, 5000);

     // show list of found devices
     WriteLn('The following devices were found:');
     for i := 0 to (PeripheralListLen - 1) do
     begin
       Peripheral := PeripheralList[i];
       PeripheralIdentifier := simpleble_peripheral_identifier(Peripheral);
       PeripheralAddress := simpleble_peripheral_address(Peripheral);
       WriteLn('[' + IntToStr(i) + '] ' + PeripheralIdentifier + ' [' + PeripheralAddress + ']');
       simpleble_free(@PeripheralIdentifier);
       simpleble_free(@PeripheralAddress);
     end;

     // select a device to connect to
     Selection := -1;
     write('Please select a device to connect to: (or -1 to quit) ');
     ReadLn(Selection);
     if (Selection < 0) or (Selection >= PeripheralListLen) then
     begin
       WriteLn('Invalid selection.');
       simpleble_adapter_release_handle(Adapter);
       break;
     end;

     Peripheral := PeripheralList[Selection];

     // connect to device
     PeripheralIdentifier := simpleble_peripheral_identifier(Peripheral);
     PeripheralAddress := simpleble_peripheral_address(Peripheral);
     WriteLn();
     WriteLn('Connecting to ' + PeripheralIdentifier + ' [' + PeripheralAddress + ']');
     simpleble_free(@PeripheralIdentifier);
     simpleble_free(@PeripheralAddress);

     ErrCode := simpleble_peripheral_connect(Peripheral);
     if ErrCode <> SIMPLEBLE_SUCCESS then
     begin
       WriteLn('Failed to connect.');
       break;
     end;
     ServicesCount := simpleble_peripheral_services_count(Peripheral);
     WriteLn('Successfully connected, listing ' + IntToStr(ServicesCount) + ' services.');
     WriteLn;

     // show gatt table with services and characteristics
     for i := 0 to (ServicesCount - 1) do
     begin
       ErrCode := simpleble_peripheral_services_get(Peripheral, i, Service);
       if ErrCode <> SIMPLEBLE_SUCCESS then
       begin
         WriteLn('Failed to get service.');
         break;
       end;
       WriteLn('Service: ' + Service.Uuid.Value + ' - (' + IntToStr(Service.CharacteristicCount) + ')');
       for j := 0 to (Service.CharacteristicCount-1) do
       begin
         WriteLn('  Characteristic: ' + Service.Characteristics[j].Uuid.Value + ' - (' + IntToStr(Service.Characteristics[j].DescriptorCount) + ')');
         for k := 0 to (Service.Characteristics[j].DescriptorCount - 1) do
           WriteLn('    Descriptor: ' + Service.Characteristics[j].Descriptors[k].Uuid.Value);
       end;
     end;

     PeripheralIdentifier := simpleble_peripheral_identifier(Peripheral);
     WriteLn('ID: '+PeripheralIdentifier);
     simpleble_free(@PeripheralIdentifier);

     // and disconnect again from device
     simpleble_peripheral_disconnect(Peripheral);

     // wait for enter key
     WriteLn();
     WriteLn('Press any key');
     ReadLn();

     // release the BLE handle
     simpleble_adapter_release_handle(Adapter);

  end;

  // stop program loop
  Terminate;
end;
```
But if I move the cnt loop so that it starts after I select the device (and the end of the loop to before the adapter release) then it fails to reconnect every time.

felixdollack, 2023-01-21:
Ah, you are releasing the adapter. I was thinking about that.
Makes somewhat sense that windows cleans up after that,
but it is the worst option fo user experience and I would like to avoid having to do that :/

Gnome16, 2023-01-21:
I only said it works. I didn't say it was pretty!!!

closed 2023-01-23 by kdewald