simpleble/simpleble #134: Expose advertising service data

Symbitic, 2022-11-16:
BLE advertises Service Data as well as manufacturer data. I would like to request exposing that. It's needed for things like Eddystone.

Some links that may help:
https://github.com/chromium/chromium/blob/main/device/bluetooth/bluetooth_adapter_mac.mm#L670
https://github.com/chromium/chromium/blob/main/device/bluetooth/bluetooth_adapter_winrt.cc#L252
https://github.com/chromium/chromium/blob/main/device/bluetooth/bluez/bluetooth_adapter_bluez.cc#L885

eriklins, 2023-02-22:
Just played with the new functions and they work pretty well. Thanks for implementing it.

Wondering if it would have been more consistent to just add data_length and data[27] to simpleble_service_t instead of adding a new data type and new functions.

Symbitic, 2023-02-22:
Actually, I was trying to be consistent with Manufacturer Data. Every platform API has almost identical APIs for SD and MD.

eriklins, 2023-02-22:
Well, yes, functions for service data and manufacturer data is almost identical, but we now have the previous service functions kind of beside the new ones (and there are not such functions existing - and not needed - for manufacturer data).

Example: I'm writing a GUI application to scan for BLE devices and display the information on a form window. Displaying of services and their uuids involve simpleble_peripheral_services_count and simpleble_peripheral_services_get. If I now would want to display service data along with the uuids from simpleble_peripheral_services_get, there is no direct link existing to the new service data functions like simpleble_peripheral_services_get . The only way to map service data to the services from simpleble_peripheral_services_get would be to loop over all simpleble_peripheral_service_data_get and compare the uuid arrays.

That's (maybe there are further reasons existing) why I would prefer to have the service and service data related functions consistent - even though they might have a slightly different API than the manufacturer data functions. My 2 cents...

kdewald, 2023-02-22:
I think @eriklins has a point. I'll make that change before the official v0.6 release.

Symbitic, 2023-02-22:
Careful. Some of the APIs can return UUIDs that weren't originally found. That's part of why I wanted it: it means there's an additional way of finding services that weren't originally seen.

eriklins, 2023-02-22:
How does that happen? You mean when we get additional services after connecting and reading the gatt table? That would be true, the gatt table can hold way more service uuids than the advert payload. However, that was already the case with the existing functions and when callling simpleble_peripheral_services_count after a connection it will no longer return the services from advertising. An application needs to handle this anyway.

kdewald, 2023-02-22:
I do like the simpler interface, and I don't think we'll have much conflict between modes.

Based on this, the new behavior would be:
- All services seen during advertisement will be accessed from the same function (`services`), the ones having data will have that information exposed as well.
- Once a connection is established, the list of advertised services gets replaced by the GATT services presented by the device

kdewald, 2023-02-22:
Correct, but I will emphasize this more thoroughly throughout the code.

This conversation was great, I just found a bug related to the previous service discovery on MacOS :D

closed 2023-02-23 by kdewald