simpleble/simpleble #135: Bluez doesn't detect services before connecting for all devices

Symbitic, 2022-11-19:
Bluez doesn't detect advertised services for every device.

When I run the SimpleBLE `scan` example on Windows, it lists plenty of devices that have services advertised (including the LEGO Hub I am using).
On Linux, when I run that same example, while there are a few devices that have services advertised (`services()`), the LEGO hub I am using is listed as having 0 services advertised.

I confirmed that `device->service_data().size()` in the simplebluez scan example returns 0 for the LEGO hub, even though there are a few other devices that do have services advertised (like a Hue lamp).

kdewald, 2023-03-20:
Hey @Symbitic could you please check if this is still happening on Linux? There's a change it might have gotten fixed with some of the changes of those portions of the code.

Symbitic, 2023-04-02:
Sadly, this is still present. When I run the following code, the LEGO Hub I'm using always returns a service_data size of 0.
```c++
adapter->set_on_device_updated([](std::shared_ptr<SimpleBluez::Device> device) {
        if (device->name() != "HUB NO.4") {
            return;
        }
        auto service_data = device->service_data();
        std::cout << "\tServices: " << service_data.size() << "\n";
});
```

closed 2023-12-01 by kdewald