simpleble/simpleble #138: CMake build fails: error C3861: wait_for_completed: identifier not found

lairdrt, 2022-11-26:
```
cmake .. -DSIMPLEBLE_LOG_LEVEL=VERBOSE
-- Building for: Visual Studio 15 2017
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.19045.
-- The CXX compiler identification is MSVC 19.16.27048.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Module support is disabled.
-- Version: 8.1.1
-- Build type:
-- CXX_STANDARD: 11
-- Performing Test has_std_11_flag
-- Performing Test has_std_11_flag - Success
-- Performing Test has_std_0x_flag
-- Performing Test has_std_0x_flag - Failed
-- Performing Test SUPPORTS_USER_DEFINED_LITERALS
-- Performing Test SUPPORTS_USER_DEFINED_LITERALS - Success
-- Performing Test FMT_HAS_VARIANT
-- Performing Test FMT_HAS_VARIANT - Success
-- Required features: cxx_variadic_templates
-- Looking for _strtod_l
-- Looking for _strtod_l - found
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Failed
-- Performing Test COMPILER_HAS_DEPRECATED
-- Performing Test COMPILER_HAS_DEPRECATED - Success
-- Windows Host Detected
-- Applying definition: /D_WIN32_WINNT=0x0A00
-- Applying definition: /D_USE_MATH_DEFINES
-- Applying private compile option: /Gd
-- Applying private compile option: /WX
-- Applying private compile option: /W1
-- Applying definition: /D_WIN32_WINNT=0x0A00
-- Applying definition: /D_USE_MATH_DEFINES
-- Applying private compile option: /Gd
-- Applying private compile option: /WX
-- Applying private compile option: /W1
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    SIMPLEBLE_LOG_LEVEL

-- Build files have been written to: C:/Users/laird/Documents/My Projects/SimpleBLE-main/simpleble/build

C:\Users\laird\Documents\My Projects\SimpleBLE-main\simpleble\build>cmake --build . -j7
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/Users/laird/Documents/My Projects/SimpleBLE-main/simpleble/build/_deps/fmt-src/CMakeLists.txt
  Building Custom Rule C:/Users/laird/Documents/My Projects/SimpleBLE-main/simpleble/CMakeLists.txt
  format.cc
  Adapter.cpp
  os.cc
  Generating Code...
  fmt.vcxproj -> C:\Users\laird\Documents\My Projects\SimpleBLE-main\simpleble\build\lib\Debug\fmtd.lib
  Peripheral.cpp
  Service.cpp
  Characteristic.cpp
  Descriptor.cpp
  ServiceBase.cpp
  CharacteristicBase.cpp
  DescriptorBase.cpp
  Exceptions.cpp
  Logging.cpp
  AdapterSafe.cpp
  PeripheralSafe.cpp
  AdapterBuilder.cpp
  PeripheralBuilder.cpp
  ServiceBuilder.cpp
  CharacteristicBuilder.cpp
  DescriptorBuilder.cpp
  AdapterBase.cpp
c:\users\laird\documents\my projects\simpleble-main\simpleble\src\backends\windows\Utils.h(34): error C3861: 'wait_for_completed': identifier not found [C:\U
sers\laird\Documents\My Projects\SimpleBLE-main\simpleble\build\simpleble.vcxproj]
  C:\Users\laird\Documents\My Projects\SimpleBLE-main\simpleble\src\backends\windows\AdapterBase.cpp(24): note: see reference to function template instantiat
  ion 'auto SimpleBLE::async_get<winrt::Windows::Foundation::IAsyncOperation<winrt::Windows::Devices::Bluetooth::BluetoothAdapter>>(const async_t &)' being c
  ompiled
          with
          [
              async_t=winrt::Windows::Foundation::IAsyncOperation<winrt::Windows::Devices::Bluetooth::BluetoothAdapter>
          ]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(10615): note: see reference to class template instantiation 'winrt::com_p
  tr<winrt::impl::IContextCallback>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(10349): note: see reference to class template instantiation 'winrt::com_p
  tr<winrt::impl::IServerSecurity>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(10308): note: see reference to class template instantiation 'std::chrono:
  :time_point<winrt::clock,winrt::Windows::Foundation::TimeSpan>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(6462): note: see reference to class template instantiation 'winrt::com_pt
  r<winrt::impl::IMarshal>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(210): note: see reference to class template instantiation 'std::array<uin
  t8_t,8>' being compiled
  PeripheralBase.cpp
c:\users\laird\documents\my projects\simpleble-main\simpleble\src\backends\windows\Utils.h(34): error C3861: 'wait_for_completed': identifier not found [C:\U
sers\laird\Documents\My Projects\SimpleBLE-main\simpleble\build\simpleble.vcxproj]
  C:\Users\laird\Documents\My Projects\SimpleBLE-main\simpleble\src\backends\windows\PeripheralBase.cpp(33): note: see reference to function template instant
  iation 'auto SimpleBLE::async_get<winrt::Windows::Foundation::IAsyncOperation<winrt::Windows::Devices::Bluetooth::BluetoothLEDevice>>(const async_t &)' bei
  ng compiled
          with
          [
              async_t=winrt::Windows::Foundation::IAsyncOperation<winrt::Windows::Devices::Bluetooth::BluetoothLEDevice>
          ]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(10615): note: see reference to class template instantiation 'winrt::com_p
  tr<winrt::impl::IContextCallback>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(10349): note: see reference to class template instantiation 'winrt::com_p
  tr<winrt::impl::IServerSecurity>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(10308): note: see reference to class template instantiation 'std::chrono:
  :time_point<winrt::clock,winrt::Windows::Foundation::TimeSpan>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(6462): note: see reference to class template instantiation 'winrt::com_pt
  r<winrt::impl::IMarshal>' being compiled
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.17763.0\cppwinrt\winrt/base.h(210): note: see reference to class template instantiation 'std::array<uin
  t8_t,8>' being compiled
  Generating Code...
```

lairdrt, 2022-11-26:
I see now that the MS Windows SDK **must be* Version 10.0.19041.0 and perhaps not higher?

kdewald, 2022-11-27:
Hey @lairdrt, thanks for reaching out.

You're correct, the Windows SDK version needs to be that one or higher, as `wait_for_completed` is an internal function provided by it. You should be able to add a newer version from the Visual Studio Installer if necessary.

Let me know if you keep having problems and we'll dive deeper if necessary.

lairdrt, 2022-11-27:
Thank you, got past this issue by installing SDK 10.0.19041.0. I can now properly build using these commands:
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_STANDARD=17
cmake --build . -j7 --config Release
cmake --install .
```
Now I have another problem in my MS Visual Studio (Community) 2022 solution (C++/WinRT) regarding mismatches.
```
1>simpleble.lib(Adapter.obj) : error LNK2038: mismatch detected for 'C++/WinRT version': value '2.0.220110.5' doesn't match value '2.0.220531.1' in WinMain.obj
1>simpleble.lib(AdapterBase.obj) : error LNK2038: mismatch detected for 'C++/WinRT version': value '2.0.220110.5' doesn't match value '2.0.220531.1' in WinMain.obj
1>simpleble.lib(PeripheralBase.obj) : error LNK2038: mismatch detected for 'C++/WinRT version': value '2.0.220110.5' doesn't match value '2.0.220531.1' in WinMain.obj
1>simpleble.lib(Utils.cpp.obj) : error LNK2038: mismatch detected for 'C++/WinRT version': value '2.0.220110.5' doesn't match value '2.0.220531.1' in WinMain.obj
1>C:\Users\laird\Documents\Visual Studio 2022\Projects\wpable-client\x64\Release\wpable-client.exe : fatal error LNK1319: 4 mismatches detected
1>Done building project "wpable-client.vcxproj" -- FAILED.
```

I have no idea where/how the C++/WinRT versions are set so I have no idea how to go about solving this one?

kdewald, 2022-11-28:
What's probably happening is that the version of the SDK that you used for SimpleBLE does not match the one of the rest of the application. There should be a setting somewhere in your project configuration to change this.

Maybe this helps: https://learn.microsoft.com/en-us/cpp/windows/how-to-use-the-windows-10-sdk-in-a-windows-desktop-application?view=msvc-170

Alternatively, you can build SimpleBLE with newer SDK versions until you find the one that matches your OS :P

closed 2022-11-28 by kdewald