simpleble/simpleble #139: CMake for Windows does not work for Release build

lairdrt, 2022-11-26:
Running the command:

`cmake .. -DSIMPLEBLE_LOCAL=ON -DCMAKE_BUILD_TYPE=Release`

Results in an empty `Debug` directory and a `lib` directory with a `Debug` sub-directory. The `simpleble.obj` file that is needed for linking in MS Visual Studio is missing from the `Debug` directory.

Running the command:

`cmake --install .`

Results in:
```
-- Install configuration: "Release"
CMake Error at cmake_install.cmake:44 (file):
  file INSTALL cannot find "C:/Users/laird/Documents/My
  Projects/SimpleBLE-main/simpleble/build/lib/Release/simpleble.lib": No
  error.
```

The commands given here:

https://simpleble.readthedocs.io/en/latest/simpleble/usage.html

Basically do not work:

1. -DSIMPLEBLE_LOG_LEVEL=DEBUG is apparently ignored (symbol never used).
2.  There is no `sudo` command under Windows (command shell).

Even when I brute-force the build process to succeed an do manage to get the install to work, the lib file dir does not have the `.obj` files needed for linking, or perhaps the install is simply not transferring the correct files.

lairdrt, 2022-11-29:
This does appear to work under Windows 10 using Windows SDK version 10.0.20348.0:

`cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_STANDARD=17 -DCMAKE_SYSTEM_VERSION=10.0.20348.0`

Windows SDK versions 10.0.22621.755 and 10.0.22000.194 target Windows 11 and both cause C++/WinRT version mismatch errors during link of MS Visual Studio Community 2022 solution build.

klaff, 2022-12-19:
In Windows 11, I found I needed to use

```
cmake .. -DSIMPLEBLE_LOG_LEVEL="[VERBOSE|DEBUG|INFO|WARNING|ERROR|FATAL]"
```

(note added quotes)

klaff, 2022-12-19:
And replace `sudo` with run `cmake --install` from a console with admin privileges.

kdewald, 2023-01-22:
@lairdrt I've just updated the documentation to be more clear about how to use the cmake installation process.

closed 2023-01-22 by kdewald