simpleble/simpleble #140: Raspberry: Could not find a version that satisfies the requirement simplepyble

AndreMiras, 2022-11-27:
It doesn't seem possible to install it for Raspberry pi.
I haven't investigated, but I assume the prebuilt binary are not provided for the arch/platform.
See the install session below for more:
```
pi@raspberrypi:~$ venv/bin/pip install simplepyble
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting simplepyble
  Could not find a version that satisfies the requirement simplepyble (from versions: )
No matching distribution found for simplepyble
pi@raspberrypi:~$ venv/bin/python --version
Python 3.7.3
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.19.50+ #1 Tue Nov 26 01:49:16 CET 2019 armv6l GNU/Linux
pi@raspberrypi:~$
```
Luckily it seems like the repo already uses `cibuildwheel` for deploying binaries to PyPI.
So maybe the fix is as easy as adding the flag for Linux ARM?
Note that later the wheels can be tested using Docker + QEMU with something like:
https://github.com/lukechilds/dockerpi

kdewald, 2022-11-28:
Hey @AndreMiras, thanks for the report!

Indeed, I haven't yet released Python bindings for RPi, but you should be able to build them locally following [these instructions](https://simpleble.readthedocs.io/en/latest/simplepyble/usage.html#installing-locally).

Let me know if you need any help!

AndreMiras, 2022-11-28:
Hi @kdewald, thanks for getting back to me.
Yes building locally should work in theory indeed.
Even though I had an issue trying to build the 0.5.0 release, see below:
```
(venv) pi@raspberrypi:~/SimpleBLE-0.5.0/simplepyble$ pip install .                                                    
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/SimpleBLE-0.5.0/simplepyble            
  Installing build dependencies ... done                                                                              
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-wmi_w7h5/setup.py", line 15, in <module>
        version_str = (root / "VERSION").read_text(encoding="utf-8").strip()
      File "/usr/lib/python3.7/pathlib.py", line 1199, in read_text
        with self.open(mode='r', encoding=encoding, errors=errors) as f:
      File "/usr/lib/python3.7/pathlib.py", line 1186, in open
        opener=self._opener)                               
      File "/usr/lib/python3.7/pathlib.py", line 1039, in _opener
        return self._accessor.open(self, flags, mode)
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/VERSION'
                                                           
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-wmi_w7h5/
(venv) pi@raspberrypi:~/SimpleBLE-0.5.0/simplepyble$
```
But indeed ideally we could provide prebuilt wheels for this arch too.

To give a bit of context, in my case simplepyble is a [dependency of the library](https://github.com/AndreMiras/edilkamin.py/blob/1.0.0/setup.py#L34) that is used in Home Assistant.
And Home Assistant is often used with Raspberry Pi which install dependencies from PyPI and alternative index servers, but doesn't seem to provide users with an easy way to compile dependencies.

javl, 2023-02-12:
In case other are trying to install simplepyble on Raspberry pi: running `pip install .` from the `simplepyble` directory as seen in the docs will throw the above error (`no such file: /tmp/VERSION`).

Try running `pip install simplepyble` from the `SimpleBLE` directory instead. This worked for me:

    sudo apt-get install libdbus-1-dev
    git clone git@github.com:OpenBluetoothToolbox/SimpleBLE.git
    cd SimpleBLE
    pip install -r simplepyble/requirements.txt
    pip install simplepyble

kdewald, 2023-02-13:
There was a comment here that I think can solve this problem:

```
In case other are trying to install simplepyble on Raspberry pi: running pip install . from the simplepyble directory as seen in the docs will throw the above error (no such file: /tmp/VERSION).

Try running pip install simplepyble from the SimpleBLE directory instead. This worked for me:

sudo apt-get install libdbus-1-dev
git clone git@github.com:OpenBluetoothToolbox/SimpleBLE.git
cd SimpleBLE
pip install -r simplepyble/requirements.txt
pip install simplepyble
```

javl, 2023-02-13:
@kdewald I posted that comment but removed it once I realized what it did: it doesn't install locally, the final ```pip install simplepyble``` command will just pull simplepyble from pypi ;)

I'm trying to build it locally so I can use my updated version that adds `peripheral.mtu()` to the python wrapper but I keep running into problems during compilation:

```
[1/67] Building CXX object CMakeFiles/_simplepyble.dir/src/wrap_adapter.cpp.o
  FAILED: CMakeFiles/_simplepyble.dir/src/wrap_adapter.cpp.o
  /usr/bin/c++ -DSIMPLEPYBLE_VERSION=\"0.6.0.dev4\" -D_simplepyble_EXPORTS -I/home/javl/SimpleBLE/simpleble/include -isystem /tmp/pip-build-env-db6fn0uj/overlay/lib/python3.9/site-packages/pybind11/include -isystem /usr/include/python3.9 -isystem /home/javl/SimpleBLE/simplepyble/_skbuild/linux-aarch64-3.9/cmake-build/simpleble/export -O3 -DNDEBUG -fPIC -fvisibility=hidden -flto -fno-fat-lto-objects -std=c++17 -MD -MT CMakeFiles/_simplepyble.dir/src/wrap_adapter.cpp.o -MF CMakeFiles/_simplepyble.dir/src/wrap_adapter.cpp.o.d -o CMakeFiles/_simplepyble.dir/src/wrap_adapter.cpp.o -c /home/javl/SimpleBLE/simplepyble/src/wrap_adapter.cpp
  c++: fatal error: Killed signal terminated program cc1plus
  compilation terminated.
```

Followed by similar errors. I'm not sure why, this seems to be the most concrete of them:

```
/tmp/pip-build-env-db6fn0uj/overlay/lib/python3.9/site-packages/pybind11/include/pybind11/pybind11.h:208:28: internal compiler error: Illegal instruction
    208 |             rec->data[0] = new capture{std::forward<Func>(f)};
        |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

Finally resulting in:

    ERROR: Could not build wheels for simplepyble, which is required to install pyproject.toml-based projects

AndreMiras, 2023-02-13:
Thanks I took a look, but same story, see the logs below:
```
(venv) pi@raspberrypi:~/SimpleBLE-0.5.0/simplepyble$ sudo apt install --yes libdbus-1-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libdbus-1-dev is already the newest version (1.12.24-0+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 184 not upgraded.
(venv) pi@raspberrypi:~/SimpleBLE-0.5.0/simplepyble$ pip install .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/SimpleBLE-0.5.0/simplepyble
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-4nol5edk/setup.py", line 15, in <module>
        version_str = (root / "VERSION").read_text(encoding="utf-8").strip()
      File "/usr/lib/python3.7/pathlib.py", line 1199, in read_text
        with self.open(mode='r', encoding=encoding, errors=errors) as f:
      File "/usr/lib/python3.7/pathlib.py", line 1186, in open
        opener=self._opener)
      File "/usr/lib/python3.7/pathlib.py", line 1039, in _opener
        return self._accessor.open(self, flags, mode)
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/VERSION'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-4nol5edk/
(venv) pi@raspberrypi:~/SimpleBLE-0.5.0/simplepyble$ 
```

javl, 2023-02-13:
@AndreMiras The version of my comment posted by kdewald missed one detail: just create that `VERSION` fiel yourself:

    echo "0.5.0" > /tmp/VERSION

AndreMiras, 2023-02-13:
Yes I'm convinced that would certainly work even though that doesn't solve the original issue which is about distributing the prebuilt wheels.
None the less the `VERSION` bug is also another valid bug and could be fixed directly in the repo.
It's not clear to me however why that VERSION file is an issue only with Raspberry Pi. I've tried to reproduce it within docker and Python 3.7 on my host with no success (`docker run -it --rm python:3.7 bash`)

kdewald, 2023-03-14:
@AndreMiras, since v0.6.1 we'll include source distributions, which will allow SimplePyBLE to be consumed from Raspbian, albeit not in pre-built form (yet).

https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/221

closed 2023-03-14 by kdewald