simpleble/simpleble #143: Duplicated callbacks after multiple notify/unsubscribe attempts

kdewald, 2022-12-02:
Bug description from Discord:

```
I'm using simplepyble under Windows and I have a question regarding repeated usage of notify and unsubscribe. In some cases, if I connect, then use notify, things work, but if I unsubscribe to that same characteristic and call notify again (without even disconnecting), I appear to have two callbacks installed and I get duplicated data. What is the expected behavior or what would the recommended pattern be if one wants to  repeatedly turn a notification on/off while otherwise maintaining a connection?
```

klaff, 2022-12-02:
@kdewald do you have any tips for how to create an MWE that would be useful for the developers? I'm seeing this with a proprietary device so I can't simply share code.

kdewald, 2022-12-02:
Given what you described, I don't think I'll need a code sample, but I might reach out if I fail to replicate the problem.

felixdollack, 2022-12-21:
I just stumbled about the same problem on windows (I see it as well on macOS, but the program for some reason doesn't crash with what should be invalid pointers...).
I pass a pointer as user data to the callback and after unsubscribe and a 2nd notify call with a different pointer I get calls from both the 1st and 2nd callback :(
Is there a way to clear the list of callbacks?

kdewald, 2022-12-22:
Hey @felixdollack, thanks for additional info. There is no way to clear the list of callbacks as the correct behavior here should not make this necessary. I'll try to get to this issue as soon as I can.

kdewald, 2023-01-23:
Ok, I was just able to replicate the problem, looking into it.

kdewald, 2023-01-25:
@felixdollack I just landed a fix for this issue and will make a new dev release for simplepyble soon.

closed 2023-01-25 by kdewald

felixdollack, 2023-01-26:
I can confirm the callback issue disappeared for me on Windows.
Great work 💯