simpleble/simpleble #152: Authentication key

siddux, 2022-12-22:
Hi, 

is there any option to connect to devices that need an auth key (like mi band 4)? I can connect to devices that do not require authentication, but for those that require an auth key, I have not found any option about how to send it.

kdewald, 2022-12-22:
Hey @siddux, thanks for reaching out.

I'm not entirely sure what you mean by auth key in this context. Are you talking about the out-of-band code that gets exchanged between both parties during pairing? If that's the case, for Windows and MacOS you rely on the operating system to handle that (you should see some sort of popup appearing) and in the case of Linux all pairing attempts are accepted by default.

siddux, 2022-12-23:
Yes I am talking about the pairing process. In server-based protocol it is necessary a key. For eixample I've tried to connect to a BLE de ice which requieres a key and the connection process is successful and I can get all the avaluable characteristics. But, when I tried to read any of them it reurns an error so I think I need to use the key in some way but don't know how to do with the library.

As an e ample, the following [repo](https://github.com/satcar77/miband4)  is a library to read from a mi band 4 which is a BLE device using an auth key for the pairing process. Any chance to do something similar with your library?

Thanks for your quick reply!

kdewald, 2023-01-05:
Oh, I see how this works. You're specifying the actual secret key exchanged between both devices. I'm not even sure that's supported at the operating system level (I have not seen anything like this in the docs of Windows, Linux or MacOS), but if you find some working example I can look into, that would be helpful.

Given that this doesn't seem like a commonly-used feature, I don't think it's getting implemented any time soon.

closed 2023-01-05 by kdewald