simpleble/simpleble #157: Failed to connect to device using examples/connect program on windows 10

hellozyemlya, 2023-01-04:
On windows 10 connect always gives "A method was called at an unexpected time." error when trying to connect to device.
Device is custom one, nrf based.

kdewald, 2023-01-05:
Hey @hellozyemlya, thanks for the report.

What version of the library gave you this error?

hellozyemlya, 2023-01-10:
It was latest master branch.

kdewald, 2023-01-10:
Can you try the code from this MR to see if it makes things better? https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/166

kdewald, 2023-01-17:
Hey @hellozyemlya, I'm not sure if we talked about this over Discord or if it was somebody else.

When dealing with custom firmware, it is possible (especially on older SDKs) for the firmware to cause this issue on the host. A few things have been changed recently that should make this more stable. I'll close this issue but feel free to reopen if it's still happening.

closed 2023-01-17 by kdewald