simpleble/simpleble #159: Safe methods throw exceptions

lairdrt, 2023-01-05:
From the docs:

> Safe layer
> 
> SimpleBLE::AdapterSafe and SimpleBLE::PeripheralSafe classes.
> 
> These objects wrap all SimpleBLE::Adapter and SimpleBLE::Peripheral objects and provide an interface that does not throw exceptions. Instead, it will return an std::optional<T> object if the function returns a value, or a boolean indicating whether the function succeeded if the original function did not return a value. The usage is functionally equivalent to their respective counterparts in the external layer.

I'm using the Safe layer and the peripheral.write() method is throwing an exception when the BT server fails to execute the write operation in a timely manner. Relevant parts of my code (which works fine when the server responds in a timely manner):
```
int deviceIdx = BTDeviceComboBox().SelectedIndex();
SimpleBLE::Safe::Peripheral peripheral = peripherals->at(--deviceIdx);
bool success = peripheral.connect();
std::string restartStr = "RESTART"; // already UTF-8 encoded
success = peripheral.write_request(WLANMANAGE_SVC_UUID, WLANMANAGE_CHR_RESTART, restartStr); # FAILS HERE WITH EXCEPTION
```
Exception indicates:

'The attribute request that was requested has encountered an error that was unlikely, and therefore could not be completed as requested.'

Full exception dump:
```
'ConnectMyPiToWiFi.exe' (Win32): Loaded 'C:\Windows\System32\Windows.Web.dll'. 
WiFi network parameters successfully written
onecoreuap\drivers\wdm\bluetooth\user\winrt\gatt\gattcharacteristic.cpp(1121)\Windows.Devices.Bluetooth.dll!00007FF93D5B0BE3: (caller: 00007FF93D5ADD4B) Exception(1) tid(7744) 8065000E The attribute request that was requested has encountered an error that was unlikely, and therefore could not be completed as requested.
Exception thrown at 0x00007FF96C79CD29 in ConnectMyPiToWiFi.exe: Microsoft C++ exception: wil::ResultException at memory location 0x0000006F611FE060.
Exception thrown at 0x00007FF96C79CD29 in ConnectMyPiToWiFi.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
onecoreuap\drivers\wdm\bluetooth\user\winrt\gatt\gattcharacteristic.cpp(407)\Windows.Devices.Bluetooth.dll!00007FF93D5ADE16: (caller: 00007FF93D5B7BC3) ReturnHr(7) tid(7744) 8065000E The attribute request that was requested has encountered an error that was unlikely, and therefore could not be completed as requested.
Exception thrown at 0x00007FF96C79CD29 (KernelBase.dll) in ConnectMyPiToWiFi.exe: WinRT originate error - 0x8065000E : 'The attribute request that was requested has encountered an error that was unlikely, and therefore could not be completed as requested.'.
'ConnectMyPiToWiFi.exe' (Win32): Loaded 'C:\Windows\System32\profext.dll'. 
Exception thrown at 0x00007FF96C79CD29 in ConnectMyPiToWiFi.exe: Microsoft C++ exception: winrt::hresult_error at memory location 0x0000006F60CFB438.
```
Am I using the Safe methods incorrectly? Shouldn't they be catching this exception?

samedayrules, 2023-01-16:
I believe the exceptions being thrown in my posting above were due to the fact that I had, when running under the Visual Studio debugger, "Break on exception" marked in a debug dialog box. So an exception was being thrown down deep (before the ::Safe call) and was being trapped by the debugger - it appears as though the ::Safe call is doing what it is supposed to do. Sorry about that...

kdewald, 2023-01-17:
Hey @samedayrules, thanks for the update! I'll mark this one as resolved, but let me know if it happens again.

closed 2023-01-17 by kdewald