simpleble/simpleble #165: simplepyble for mac M1 ARM architecture - simplepyble.cpython-38-darwin.so not built for arm64?

z3ugma, 2023-01-10:
After installing `SimplePyBLE` on an M1 MacBook Air, I get warned that the `.so` file is for the wrong architecture. Is there a build for ARM64 / M1 Mac?

```
%pip install simplepyble
Collecting simplepyble
  Downloading simplepyble-0.5.0-cp38-cp38-macosx_11_0_arm64.whl (550 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 550.6/550.6 kB 5.4 MB/s eta 0:00:00
Installing collected packages: simplepyble
Successfully installed simplepyble-0.5.0

% python simp.py 
Traceback (most recent call last):
  File "simp.py", line 1, in <module>
    import simplepyble
ImportError: dlopen(/Users/fred/o2r/venv/lib/python3.8/site-packages/simplepyble.cpython-38-darwin.so, 0x0002): tried: '/Users/fred/o2r/venv/lib/python3.8/site-packages/simplepyble.cpython-38-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/fred/o2r/venv/lib/python3.8/site-packages/simplepyble.cpython-38-darwin.so' (no such file), '/Users/fred/o2r/venv/lib/python3.8/site-packages/simplepyble.cpython-38-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
(venv) ?1 o2r % 
```

z3ugma, 2023-01-10:
Installing `pip install simplepyble==0.6.0.dev2` appears to work. Is the 0.6.0 release ready for production and publication to PyPI?

kdewald, 2023-01-10:
@z3ugma thanks for posting a reply! It will be out soon, there are a few things that I still want to add to it.

closed 2023-01-10 by kdewald