simpleble/simpleble #169: How about make some release binary files?

BtuApple, 2023-01-11:
I want use this, and do not edit code.Just use the binary file for my program.I you can release some binary,it will be very helpful.

kdewald, 2023-01-12:
Hey @BtuApple,

That's a good suggestion, I'll try to get that for the v0.6 release.

Gnome16, 2023-01-22:
In the meantime you can get binaries for Win32, Win64 and Linux64 from the Pascal bindings project:
[https://github.com/eriklins/Pascal-Bindings-For-SimpleBLE-Library/releases](url)
But please see issue number 12 on that project if you want the Linux ones. You might need to use patchelf to fix the RUNPATH in the SO files.

A "sudo apt-get install simpleble" option would be ideal for end users though so they don't need to worry about the files at all.

kdewald, 2023-02-11:
Feature has been added in https://github.com/OpenBluetoothToolbox/SimpleBLE/commit/23792f5c2e9067905ba56fd7c0ac4d26f62ee786 and https://github.com/OpenBluetoothToolbox/SimpleBLE/commit/8e2ec4a44aedd8664868b365880f0f90c1975533. I've manually built the artifacts for v0.5.0 and the ones for the next release will be built automatically.

closed 2023-02-11 by kdewald

BtuApple, 2023-02-11:
Got it！