simpleble/simpleble #173: Windows lib install copies files to same dir

samedayrules, 2023-01-17:
Under Windows, when I am making the library files for different configs (--config **Release** and --config **Debug**), the resulting library files are overwritten as the file destination is the same (e.g., `C:\Program Files (x86)\simpleble\lib`).

Is there a way to make the Release library go to `C:\Program Files (x86)\simpleble\lib\Release\simpleble.lib` and the Debug library go to `C:\Program Files (x86)\simpleble\lib\Debug\simpleble.lib` as you would expect with other components targeting different configurations under Windows Visual Studio?

I've tried to experiment with CMake variables such as the prefix-path, but that does not allow me to specify a path suffix of use after `\lib` in the library destination based upon varying configurations.

kdewald, 2023-01-17:
Oh man, this one has been bugging me for quite a while, but it seems to come from Visual Studio. I'll see if I can add a post-build hook to copy everything to the correct location. If you can find any information on it to help speed up the process, that would be super useful.

eriklins, 2023-01-31:
I modified the build_lib.ps1 to use different build paths for x64 and Win32 (build_simpleble_x64 and build_simpleble_Win32). I first tried just to modify the install directory inside the overall example build directory, but this does not work. When compiling first e.g., x64 and then Win32 on the second run the build system "thinks" all the object files would be up-to-date and won't compile for the new architecture. Attached is my modified build_lib.ps1. Better ideas / comments welcome... Or let me know if I should do a pull request with this. 
[build_lib.ps1.zip](https://github.com/OpenBluetoothToolbox/SimpleBLE/files/10547257/build_lib.ps1.zip)