simpleble/simpleble #177: Building example on windows fails after upgrade from 5f0ef4e to 8110d1f (master HEAD at point of writing)

felixdollack, 2023-01-24:
In an attempt to test the change merged from #157 I am creating a build folder, running `cmake ../examples/simpleble/` and attempting compilation of simpleble_examples.sln in MCVS2022.
But instead of a working binary I am greeted with the following error:

```
C:\Users\felix\Documents\SimpleBLE\build\_deps\fmt-src\include\fmt/chrono.h(1,1): error C2220: the following warning is treated as an error

C:\Users\felix\Documents\SimpleBLE\build\_deps\fmt-src\include\fmt/chrono.h(1,1): warning C4819: The file contains a character that cannot be represented in the current code page (932). Save the file in Unicode format to prevent data loss
```

felixdollack, 2023-01-24:
It still works with be8cdac, but stops in f9f7daa.

kdewald, 2023-01-25:
Hey @felixdollack, thanks for the report. Can you confirm that this also happens when you clear the entire build cache? I suspect there could be some contamination happening, given that I don't see that problem on our CI builds: https://github.com/OpenBluetoothToolbox/SimpleBLE/actions/runs/3989181453

felixdollack, 2023-01-25:
Thank you @kdewald for replying so quickly.
Yes, it happens even after calling `Build -> Clean Solution` or with a newly cloned version.
It also is the same for Debug and Release build.

felixdollack, 2023-01-26:
I found the reason.
fmt/chrono.h has Japanese Kanji characters in one of the comments, if I remove them it compiles without problem.

https://github.com/fmtlib/fmt/blob/a2c05a10ec875712bfee739a05ed99fc3915f4e9/include/fmt/chrono.h#L663

Since it is an external dependency you pull in I will open a [new issue](https://github.com/fmtlib/fmt/issues/3289) with them directly.
I also change the title of this issue to the error message in the hope others who have the same problem will find the solution (or workaround) here.

kdewald, 2023-01-26:
That's really bizarre, good catch! Do you know if setting a different version of `fmt` in `cmake/find/Findfmt.cmake` does fix it? (As long as chrono is supported I think we should be fine)

felixdollack, 2023-01-26:
Your previous version of fmtlib was 8.1.1 which was released January 2022.
The line in question is part of fmtlib since at least 2019.
So it is even more bizarre that I could build it before 😆

There is only v9.0 between the previous and current selected fit release.
I will try it, but I don't have much hope.

felixdollack, 2023-01-27:
Hey @kdewald, it seems MSVisualStupido is using ones system locale as the source character set. That is most likely responsible for the issue:
https://developercommunity.visualstudio.com/t/warning-c4819-to-file-saved-in-utf-8-encoding/176752#T-N1092308

Would it be possible to add `/utf-8` at the right places in the cmake instructions?
Something like this:
```
add_compile_options("$<$<C_COMPILER_ID:MSVC>:/utf-8>")
add_compile_options("$<$<CXX_COMPILER_ID:MSVC>:/utf-8>")
```

kdewald, 2023-01-27:
We can definitely make that change. As I don't have a machine on a different locale, could you test it and submit a PR? I'll be quick to review it.

closed 2023-01-27 by kdewald