simpleble/simpleble #186: [Windows] Building SimpleBLE for 32-bit Windows always returns "Bluetooth not enabled"

kdewald, 2023-02-01:


kdewald, 2023-02-06:
@eriklins I tried to replicate this issue locally but failed. Is there anything specific about your environment that could cause this problem?

GrahamNZ, 2023-02-10:
I built the DLL files and the samples on a Win 11 PC as 32 bit. The example_list_adapters program shows my BLE adapter correctly and the example_scan program found devices.

Then I copied the fmt.dll, simpleble.dll, simple-c.dll and example_scan.exe files to my Win 10 PC (where the 64 bit version works fine, as does the nSoftware 32 bit BLE program) and it fails with a "Bluetooth is not enabled!" error. The Win 10 PC does not have VS, Git, Cmake or C++ installed.

kdewald, 2023-02-11:
https://learn.microsoft.com/en-us/windows/win32/api/sysinfoapi/nf-sysinfoapi-getnativesysteminfo

closed 2025-06-21 by kdewald