simpleble/simpleble #189: bluetooth_enabled throws SimpleDBus::Exception::NotInitialized

jacobbreen25, 2023-02-04:
I am working on programming an application however, I cannot get started because when I run my compiled code, it says:
`terminate called after throwing an instance of 'SimpleDBus::Exception::NotInitialized'
  what():  Object not initialized.`

Here is my main.cpp:
`#include <iostream>
#include "simpleble/SimpleBLE.h"

int main(int argc, char* argv[]) {
    SimpleBLE::Adapter adpt;
    adpt.bluetooth_enabled();
    return 0;
}
`

and here is my Makefile:
`cmake_minimum_required(VERSION 3.7)

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)

project(
    test
)
include (FetchContent)
FetchContent_Declare(simpleble GIT_TAG "origin/main" GIT_REPOSITORY https://github.com/OpenBluetoothToolbox/SimpleBLE.git ) 
FetchContent_GetProperties(simpleble)
if(NOT simpleble_POPULATED)
    FetchContent_Populate(simpleble)
    list(APPEND CMAKE_MODULE_PATH "${simpleble_SOURCE_DIR}/cmake/find")
    add_subdirectory("${simpleble_SOURCE_DIR}/simpleble" "${simpleble_BINARY_DIR}")
endif()

set(simpleble_FOUND 1)

add_executable(main main.cpp)
target_link_libraries(main simpleble::simpleble)
`

kdewald, 2023-02-05:
Hey @jacobbreen25, please review the tutorial and the examples on how to use SimpleBLE.
https://simpleble.readthedocs.io/en/latest/simpleble/tutorial.html#getting-started

Thank you!

jacobbreen25, 2023-02-05:
> Hey @jacobbreen25, please review the tutorial and the examples on how to use SimpleBLE. https://simpleble.readthedocs.io/en/latest/simpleble/tutorial.html#getting-started
> 
> Thank you!

Thank you for your time,

 I have tried with your provided examples on GitHub as well as the tutorial and I get the same exception being thrown. It is always when I am trying to enable the bluetooth, no matter if I do ```SimpleBLE::Adapter::bluetooth_enabled()``` or if I do it as I specified above. Because it consistently throws this error, I cannot do even the most basic tutorial as it will not get past the first function called.
 
I am going to begin looking through the source code to see if I can't find where this exception is being thrown and why.

kdewald, 2023-02-06:
It's possible this bug relates to: https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/131

I need to properly investigate further, but my understanding is that Windows does some weird things to provide a running Bluez instance, so it's highly likely it won't work.

I'd recommend running your app directly on your native system, which definitely should work.

laborg, 2023-02-08:
I'm trying to make SimpleBLE available in Julia but stumbled upon this error. This stacktrace might help debugging the error:
```
julia> SimpleBLE.simpleble_adapter_get_count()
terminate called after throwing an instance of 'SimpleDBus::Exception::NotInitialized'
  what():  Object not initialized.

signal (6): Aborted
in expression starting at REPL[2]:1
pthread_kill at /lib/x86_64-linux-gnu/libc.so.6 (unknown line)
raise at /lib/x86_64-linux-gnu/libc.so.6 (unknown line)
abort at /lib/x86_64-linux-gnu/libc.so.6 (unknown line)
__verbose_terminate_handler at /workspace/srcdir/gcc-12.1.0/libstdc++-v3/libsupc++/vterminate.cc:95
__terminate at /workspace/srcdir/gcc-12.1.0/libstdc++-v3/libsupc++/eh_terminate.cc:48
__cxa_call_terminate at /workspace/srcdir/gcc-12.1.0/libstdc++-v3/libsupc++/eh_call.cc:54
__gxx_personality_v0 at /workspace/srcdir/gcc-12.1.0/libstdc++-v3/libsupc++/eh_personality.cc:688
_Unwind_RaiseException_Phase2 at /workspace/srcdir/gcc-12.1.0/libgcc/unwind.inc:64
_Unwind_RaiseException at /workspace/srcdir/gcc-12.1.0/libgcc/unwind.inc:136
__cxa_throw at /workspace/srcdir/gcc-12.1.0/libstdc++-v3/libsupc++/eh_throw.cc:93
_ZN10SimpleDBus10Connection12remove_matchENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE.cold.29 at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble.so (unknown line)
_ZN11SimpleBluez5BluezD2Ev at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble.so (unknown line)
_ZN9SimpleBLE5BluezC2Ev.cold.299 at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble.so (unknown line)
_ZN9SimpleBLE5Bluez3getEv at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble.so (unknown line)
_ZN9SimpleBLE11AdapterBase12get_adaptersEv at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble.so (unknown line)
_ZN9SimpleBLE7Adapter12get_adaptersEv at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble.so (unknown line)
_ZN9SimpleBLE4Safe7Adapter12get_adaptersEv at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble.so (unknown line)
simpleble_adapter_get_count at /home/aigner/.julia/artifacts/15ed377b3da6c00690a97a39fc7cc57a95354db7/lib/libsimpleble-c.so (unknown line)
simpleble_adapter_get_count at /home/aigner/code/SimpleBLE.jl/SimpleBLE.jl:11
unknown function (ip: 0x7efefc51e8bf)
_jl_invoke at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/gf.c:2377 [inlined]
ijl_apply_generic at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/gf.c:2559
jl_apply at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/julia.h:1843 [inlined]
do_call at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/interpreter.c:126
eval_value at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/interpreter.c:215
eval_stmt_value at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/interpreter.c:166 [inlined]
eval_body at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/interpreter.c:612
jl_interpret_toplevel_thunk at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/interpreter.c:750
jl_toplevel_eval_flex at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/toplevel.c:906
jl_toplevel_eval_flex at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/toplevel.c:850
jl_toplevel_eval_flex at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/toplevel.c:850
jl_toplevel_eval_flex at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/toplevel.c:850
ijl_toplevel_eval_in at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/toplevel.c:965
eval at ./boot.jl:368 [inlined]
eval_user_input at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/usr/share/julia/stdlib/v1.8/REPL/src/REPL.jl:151
repl_backend_loop at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/usr/share/julia/stdlib/v1.8/REPL/src/REPL.jl:247
start_repl_backend at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/usr/share/julia/stdlib/v1.8/REPL/src/REPL.jl:232
#run_repl#47 at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/usr/share/julia/stdlib/v1.8/REPL/src/REPL.jl:369
run_repl at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/usr/share/julia/stdlib/v1.8/REPL/src/REPL.jl:355
jfptr_run_repl_65104.clone_1 at /home/aigner/julia-1.8.5/lib/julia/sys.so (unknown line)
_jl_invoke at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/gf.c:2377 [inlined]
ijl_apply_generic at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/gf.c:2559
#967 at ./client.jl:419
jfptr_YY.967_33139.clone_1 at /home/aigner/julia-1.8.5/lib/julia/sys.so (unknown line)
_jl_invoke at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/gf.c:2377 [inlined]
ijl_apply_generic at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/gf.c:2559
jl_apply at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/julia.h:1843 [inlined]
jl_f__call_latest at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/builtins.c:774
#invokelatest#2 at ./essentials.jl:729 [inlined]
invokelatest at ./essentials.jl:726 [inlined]
run_main_repl at ./client.jl:404
exec_options at ./client.jl:318
_start at ./client.jl:522
[... cut ...]
jl_apply at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/julia.h:1843 [inlined]
true_main at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/jlapi.c:575
jl_repl_entrypoint at /cache/build/default-amdci4-2/julialang/julia-release-1-dot-8/src/jlapi.c:719
main at julia (unknown line)
unknown function (ip: 0x7eff12995d8f)
__libc_start_main at /lib/x86_64-linux-gnu/libc.so.6 (unknown line)
unknown function (ip: 0x401098)
Allocations: 2909 (Pool: 2896; Big: 13); GC: 0
[1]    5730 IOT instruction (core dumped)  julia --project=.

```

kdewald, 2023-02-09:
@laborg just to confirm, are you also using WSL?

laborg, 2023-02-09:
No, I'm on a native Ubuntu installation but the build of `simpleble` was done on Arch. If I rely on the native ubuntu provided `libdbus-1` as a dependency the build works. If I use a separately built `libdbus-1` the error appears. So I guess the question is rather how to debug why the other `libdbus-1` doesn't work. Could you give any pointers on the requirements/constraints of `libdbus-1` for the use in `simpleble`?

kdewald, 2023-02-09:
@jacobbreen25 FYI, it turns out that Bluetooth is not supported by WSL, so SimpleBLE won't work in that environment. I've created a separate issue to detect the problem so that the error becomes explicit.

https://github.com/microsoft/WSL/issues/423

kdewald, 2023-02-09:
@laborg it seems that the issue you're getting is a fatal error that probably originates from the same issue I marked in https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/196, were a condition is incorrectly handled.

On top of that, as far as I know nobody has tested SimpleBLE on Arch, so beware that things might be broken. 

In this specific case of `libdbus-1`, OSes impose some specific config options to it (the most notable I know being the path where the expected sockets should be found), so it could be possible that the client is failing to receive any packets, which I think could trigger the bug you saw above.

My advice here would be to not mix builds between distros. SimpleBLE heavily relies on OS-specific modules, so messing with them is bound to cause some kind of unexpected behavior.

kdewald, 2023-02-09:
Bingo, I was able to replicate the failure on WSL with the correct message, but I'm pretty sure your build will fail similarly.

```
Bluetooth enabled: [ERROR] SimpleBLE: /home/kdewald/Workspace/SimpleBLE/simpleble/src/frontends/base/Adapter.cpp:25 in bluetooth_enabled: Failed to check if bluetooth is enabled: org.freedesktop.DBus.Error.FileNotFound: Failed to connect to socket /run/dbus/system_bus_socket: No such file or directory
0
terminate called after throwing an instance of 'SimpleDBus::Exception::DBusException'
  what():  org.freedesktop.DBus.Error.FileNotFound: Failed to connect to socket /run/dbus/system_bus_socket: No such file or director
```

kdewald, 2023-02-09:
I'll close this issue in favor of the other one, as now the fix is clear.

closed 2023-02-09 by kdewald