simpleble/simpleble #191: [MacOS] RPath incorrectly set

kdewald, 2023-02-06:
From Discord:

```
Hi. Quick question - when I build SimpleBLE as shared library, it requires me to add DYLD_LIBRARY_PATH=. MyApp
Otherwise libraries not found.
Is there any way to make it work without DYLD_LIBRARY_PATH?
```

hellozyemlya, 2023-02-06:
The quick solution here:
`install_name_tool -add_rpath @executable_path ./lib/libsimpleble-c.dylib`(just in case, it is safe to put that for c++ variant library as well)
It works, but might be idiomatically incorrect(very unsure about it).

It will be great to add build option that will allow to execute install_name_tool optionally when building for MacOS. 
But also it seems like that is an distribution/application responsibility to resolve rpath correctly. I don't know :( 

See https://blog.krzyzanowskim.com/2018/12/05/rpath-what/, also there is plenty of stackoverflow topics and other sources over google.

kdewald, 2023-02-07:
Fixed in https://github.com/OpenBluetoothToolbox/SimpleBLE/commit/c70cb0ed5cc3857beeb55672fb982ebcb89f05ab

closed 2023-02-07 by kdewald