simpleble/simpleble #193: Connecting without scanning

GeoffreyMClark, 2023-02-07:
Is it possible to connect to a device without having to scan first, assuming you know the devices address of course? 

The scan feature does not always find my device, though I can connect through the terminal easily enough every single time.

eriklins, 2023-02-07:
That's not possible from how BLE works perspective. A BLE advertisement is kind of an invitation to engage a connection with a peripheral device. Without this, it's not possible to connect, even not if the BLE MAC address of a particular device would be known. If you have the peripheral device under your control, you could try to increase the advertisement interval to make it showing up eariler/more frequent. From the central perspective it would be possible to play with scanning parameters like scan interval and scan window, but as far as I see, it's not possible to tweak these with some SimpleBLE function. Question would go to @kdewald whether it would be possible to add this, provided the underlying Windows API would give access to these parameters.

kdewald, 2023-02-07:
Hey @GeoffreyMClark, thanks for reaching out!

As @eriklins mentioned, there is no way to connect without scanning first. Unfortunately, operating systems don't provide transparent settings to change the scan behavior, so you're basically stuck with whatever settings the OS is using.

To improve your chances of a device being picked up, you can:
- Increase the advertising frequency, as @eriklins said.
- Bring both devices closer to each other.
- Increase the scan time.

closed 2023-02-07 by kdewald