simpleble/simpleble #194: Manufacturer Specific AD Elements in Scan Callback Missing

eriklins, 2023-02-07:
I found issues with manufacturer specific data with the SimpleBLE C library on Windows:

The scan callback never exposes more than two manufacturer specific data elements. Technically it is possible to have e.g., two MD elements in the advert report and another two in the scan report. I made such a device for testing and SimpleBLE exposes just the two MDs from the advert report but none from the scan report in the scan callback event. If I have just one MD in the advert report and one in the scan report, then both are exposed in the same callback event.

For the above I used four MD elements all having different manufacturer ids. If I change the situation to just two MD elements in one report with the SAME manufacturer id, then two subsequent callback events are thrown for each advert. One contains the first MD element, the other one contains the second MD element. Not sure if this is desired and we won't actually loose any advert information, but from an application perspective it looks like the content of the AD elements would change from advert to advert instead of realising it's two different AD elements.

However, all this might be picky and in the real world more than one AD element per report is an uncommon situation. Just wanted to raise it to let others know.

kdewald, 2023-03-16:
@eriklins could you test this again with the latest release? There was a related bug on Windows, so it might already be solved.

eriklins, 2023-03-17:
@kdewald I pulled the latest main as of today and still the same. Only the first MD element is shown per advert and scan report.

closed 2023-08-09 by eriklins