simpleble/simpleble #195: Include file properties in Windows binaries

GrahamNZ, 2023-02-08:
Windows files have properties such as version, description, copyright etc that installation programs such as Inno Setup can use to ensure that new versions aren't overwritten with old ones.

kdewald, 2023-02-08:
Hey @GrahamNZ, do you know if this is supported by CMake in any way? Any pointers in that direction would be very helpful.

GrahamNZ, 2023-02-08:
I believe you need to include a version resource file (*.rc) that is used by the linker, but I'm not sure how to achieve that in CMake. I think creating the file is an option in Visual Studio?

kdewald, 2023-02-09:
It must be something that can be automated. Manual steps don't scale.

CMake should have something on this front. You can try asking ChatGPT to see if it can give you a pointer :D

GrahamNZ, 2023-02-11:
OK, here is an example on simpleble.dll but it probably needs to be done for the other files too, and I don't know enough about the structures to know if I've done it in the right place. But it should be a good starting point.

1. VS: Open ..\build_simpleble\simpleble.vcxproj
2. VS: Project > Add resource > Version
3. VS: Type in FILEVERSION as 0,6,0,0 and enter the other 'TODO' fields as required. Close.
4. Copy the new ALL_BUILD.rc file to the ..\simpleble folder
5. Edit CMakeLists.txt, adding the line:
_set(SIMPLEBLE_SRC_
       ${CMAKE_CURRENT_SOURCE_DIR}/ALL_BUILD.rc
       _${CMAKE_CURRENT_SOURCE_DIR}/src/frontends/base/Adapter.cpp_
6. Run utils\build_lib.bat [reqd parameters]

The simpleble.dll in ..\build_simpleble\bin\Release now has the file details.