simpleble/simpleble #200: simpleble_peripheral_connect() returns SIMPLEBLE_SUCCESS when trying to connect to non-existing device.

eriklins, 2023-02-11:
I used the example_connect_c.exe on Windows 64 bit, which scans for devices and then allows to connect to one of them. After the scan process I switched off my device (name XYZ) but selected to connect to it from the list. It tries to connect, takes very long and finally outputs "Successfully connected" even though the device is turned off. It then outputs some services, which actually are the services I included in the advertisment (GAP). Below is the terminal output. The first cycle is with device XYZ active and successfully connecting and listing GATT services, the second cycle is with device XYZ switched off after scanning but still showing successfully connected and listing GAP services from the previous advertisment.

d:\gitlocal\SimpleBLE_main\build_simpleble_examples\bin\Release>example_connect_c.exe
[INFO] SimpleBLE: D:\gitlocal\SimpleBLE_main\simpleble\src\backends\windows\Utils.cpp:33 in initialize_winrt: CoGetApartmentType: cotype=-1, qualifier=0, result=800401F0
[INFO] SimpleBLE: D:\gitlocal\SimpleBLE_main\simpleble\src\backends\windows\Utils.cpp:41 in initialize_winrt: RoInitialize: result=0
Adapter ERIKSDESKTOP - Oben started scanning.
Adapter ERIKSDESKTOP - Oben found device:  [fd:e9:23:0c:77:98]
Adapter ERIKSDESKTOP - Oben found device: LAIRD BL654-ECAFEA [c9:d4:85:ec:af:ea]
Adapter ERIKSDESKTOP - Oben found device:  [6a:a3:86:fb:e1:d8]
Adapter ERIKSDESKTOP - Oben found device: XYZ [fc:81:1a:0a:7d:c0]
Adapter ERIKSDESKTOP - Oben found device:  [12:a6:98:e9:9c:c8]
Adapter ERIKSDESKTOP - Oben found device:  [46:bc:f8:55:53:31]
Adapter ERIKSDESKTOP - Oben found device:  [24:74:46:56:14:42]
Adapter ERIKSDESKTOP - Oben found device:  [58:ef:49:ce:63:38]
Adapter ERIKSDESKTOP - Oben found device:  [57:77:de:f7:0d:eb]
Adapter ERIKSDESKTOP - Oben stopped scanning.
The following devices were found:
[0] BT710-7798 [fd:e9:23:0c:77:98]
[1] LAIRD BL654-ECAFEA [c9:d4:85:ec:af:ea]
[2]  [6a:a3:86:fb:e1:d8]
[3] XYZ [fc:81:1a:0a:7d:c0]
[4]  [12:a6:98:e9:9c:c8]
[5]  [46:bc:f8:55:53:31]
[6]  [24:74:46:56:14:42]
[7]  [58:ef:49:ce:63:38]
[8]  [57:77:de:f7:0d:eb]
Please select a device to connect to: 3
Connecting to XYZ [fc:81:1a:0a:7d:c0]
Successfully connected, listing 3 services.
Service: 00001800-0000-1000-8000-00805f9b34fb - (4 characteristics)
  Characteristic: 00002a00-0000-1000-8000-00805f9b34fb - (0 descriptors)
  Characteristic: 00002a01-0000-1000-8000-00805f9b34fb - (0 descriptors)
  Characteristic: 00002a04-0000-1000-8000-00805f9b34fb - (0 descriptors)
  Characteristic: 00002aa6-0000-1000-8000-00805f9b34fb - (0 descriptors)
Service: 00001801-0000-1000-8000-00805f9b34fb - (1 characteristics)
  Characteristic: 00002a05-0000-1000-8000-00805f9b34fb - (1 descriptors)
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb
Service: f5251000-5b28-45f6-b8c8-d5e5229d2273 - (6 characteristics)
  Characteristic: f5251001-5b28-45f6-b8c8-d5e5229d2273 - (0 descriptors)
  Characteristic: f5251002-5b28-45f6-b8c8-d5e5229d2273 - (1 descriptors)
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb
  Characteristic: f5251003-5b28-45f6-b8c8-d5e5229d2273 - (0 descriptors)
  Characteristic: f5251004-5b28-45f6-b8c8-d5e5229d2273 - (0 descriptors)
  Characteristic: f5251005-5b28-45f6-b8c8-d5e5229d2273 - (0 descriptors)
  Characteristic: f5251006-5b28-45f6-b8c8-d5e5229d2273 - (1 descriptors)
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb
Releasing allocated resources.

d:\gitlocal\SimpleBLE_main\build_simpleble_examples\bin\Release>
d:\gitlocal\SimpleBLE_main\build_simpleble_examples\bin\Release>
d:\gitlocal\SimpleBLE_main\build_simpleble_examples\bin\Release>example_connect_c.exe
[INFO] SimpleBLE: D:\gitlocal\SimpleBLE_main\simpleble\src\backends\windows\Utils.cpp:33 in initialize_winrt: CoGetApartmentType: cotype=-1, qualifier=0, result=800401F0
[INFO] SimpleBLE: D:\gitlocal\SimpleBLE_main\simpleble\src\backends\windows\Utils.cpp:41 in initialize_winrt: RoInitialize: result=0
Adapter ERIKSDESKTOP - Oben started scanning.
Adapter ERIKSDESKTOP - Oben found device:  [57:77:de:f7:0d:eb]
Adapter ERIKSDESKTOP - Oben found device:  [46:bc:f8:55:53:31]
Adapter ERIKSDESKTOP - Oben found device:  [58:ef:49:ce:63:38]
Adapter ERIKSDESKTOP - Oben found device: LAIRD BL654-ECAFEA [c9:d4:85:ec:af:ea]
Adapter ERIKSDESKTOP - Oben found device:  [12:a6:98:e9:9c:c8]
Adapter ERIKSDESKTOP - Oben found device: XYZ [fc:81:1a:0a:7d:c0]
Adapter ERIKSDESKTOP - Oben found device:  [ce:7f:8e:2e:39:2d]
Adapter ERIKSDESKTOP - Oben found device:  [24:74:46:56:14:42]
Adapter ERIKSDESKTOP - Oben found device:  [6a:a3:86:fb:e1:d8]
Adapter ERIKSDESKTOP - Oben stopped scanning.
The following devices were found:
[0]  [57:77:de:f7:0d:eb]
[1]  [46:bc:f8:55:53:31]
[2]  [58:ef:49:ce:63:38]
[3] LAIRD BL654-ECAFEA [c9:d4:85:ec:af:ea]
[4]  [12:a6:98:e9:9c:c8]
[5] XYZ [fc:81:1a:0a:7d:c0]
[6]  [ce:7f:8e:2e:39:2d]
[7]  [24:74:46:56:14:42]
[8]  [6a:a3:86:fb:e1:d8]
Please select a device to connect to: 5
Connecting to XYZ [fc:81:1a:0a:7d:c0]
Successfully connected, listing 2 services.
Service: 00000201-0000-1000-8000-00805f9b34fb - (0 characteristics)
Service: 00000403-0000-1000-8000-00805f9b34fb - (0 characteristics)
Releasing allocated resources.

d:\gitlocal\SimpleBLE_main\build_simpleble_examples\bin\Release>

eriklins, 2023-02-13:
Just want to add that when connecting to a device fails for some other reason, then simpleble_peripheral_connect() returns SIMPLEBLE_FAILURE properly. The above only happens, in case the device is no longer existing.

kdewald, 2023-02-25:
Synced on Discord, turns out there wasn't a bug in the library.

eriklins, 2023-02-25:
Sorry Kevin, but on discord I was talking about simpleble_peripheral_is_connected() which is working properly. This issue here was on simpleble_peripheral_connect() which does return SIMPLEBLE_SUCCESS when trying to connect to a device found during scanning but disabled/turned off prior to connecting. Issue should be re-opened, sorry.

kdewald, 2023-02-25:
My bad, reopening.

kdewald, 2023-03-16:
Found the bug, no exception is thrown if a connection is failed to be established
[image]

The fix will be up shortly.

kdewald, 2023-03-16:
Fixed in https://github.com/OpenBluetoothToolbox/SimpleBLE/commit/c858ef04f4615b9da08806c6881c6fad3010c942

closed 2023-03-16 by kdewald