simpleble/simpleble #206: Expose peripheral handle in notification and indication callback functions

eriklins, 2023-02-14:
I'm working on a (graphical) application which scans for adverts and can connect to one or more devices which are displayed in one or more windows/dialogs, showing the gatt structure and allowing to read, write and subscribe to notification/indications etc. Since I don't know about the gatt structure and how many characteristics of each device support notifications (or indications), I (need to) use one callback for all subscriptions. This is not an issue when having just one device since the service and characteristic uuids exposed in the callback can be considered unique. But this is no longer the case when connecting to two similar devices simultaneously which expose the same gatt structure and uuids. It's no longer possible to determine which device gave the notification from just the service and characteristic uuid in the callback. So, would be great (necessary) to have access to the peripheral handle in the notification and indication callback, similar to the other callbacks. Thanks!

kdewald, 2023-02-15:
That's a great idea!

eriklins, 2023-08-04:
Did you have a chance to look into this...?

KlemenDEV, 2023-12-20:
Couldn't the userdata be used to pass the peripheralHandle?

eriklins, 2024-02-21:
Hello @kdewald , just wanted to emphasize this missing feature, since it definitely limits the use of SimpleBLE for any application where we would want to connect and interact with identical devices exposing the same GATT table configuration. It's not possible to identify which device did throw a notification or indication based on what's currently exposed in the callback functions. We just get the service UUID and characteristic UUID and those would be the same for devices with same GATT configuration. Exposing the mac address as an additional parameter (or at least in the *userdata) would solve this. Or exposing the peripheral handle would the job as well.

darshan-slx, 2024-04-19:
Agreed with @eriklins.

@kdewald I am facing a similar issue of peripheral handle not being exposed in callback functions of Peripheral class.

`void set_callback_on_connected(std::function<void()> on_connected)`
`void set_callback_on_disconnected(std::function<void()> on_disconnected)`

It would be great if peripheral handle was exposed for the above two functions along with others mentioned by @eriklins.

kdewald, 2024-04-19:
Hey all, thank you for bringing this up.

This is part of our current batch of features, so we are actively going to pursue this soon. Please hang tight for a little bit longer :D

eriklins, 2024-04-19:
@darshan-slx

I am using the C version of SimpleBLE and the peripheral handle is exposed into the callback functions:

`SIMPLEBLE_EXPORT simpleble_err_t simpleble_peripheral_set_callback_on_connected(simpleble_peripheral_t handle, void (*callback)(simpleble_peripheral_t peripheral, void* userdata), void* userdata);`

Not sure about the cpp version, though.

darshan-slx, 2024-04-19:
> Hey all, thank you for bringing this up.
> 
> This is part of our current batch of features, so we are actively going to pursue this soon. Please hang tight for a little bit longer :D

Thank you!

eriklins, 2024-07-28:
Any chance to look into this enhancement? Still not possible to subscribe notifications/indications when connected to multiple devices exposing an identical gatt table. Would „just“ need to expose the peripheral handle into the callback functions either as additional parameter or into the user data. Thanks!

kdewald, 2024-07-29:
I shortlisted it for the v0.8.x release. I need to finish a contract for a client and I'll see if I can sneak this in right now.

kdewald, 2024-07-29:
Implemented in the most recent commit:
https://github.com/OpenBluetoothToolbox/SimpleBLE/commit/96038864f454793b2c7f135dbd1885aa60eba765

closed 2024-07-29 by kdewald