simpleble/simpleble #208: Static and shared build releases are equal

laborg, 2023-02-23:
First off, thanks for publishing SimpleBLE as shared lib.
I was looking forward to use them to create Julia bindings, but it turns out that they are equal to the static lib versions.
In the CI script 
https://github.com/OpenBluetoothToolbox/SimpleBLE/blob/a70c6e6f72199e402eb3d8a9e2df103c33151f9d/.github/workflows/ci_cpp_release.yml#L52
you specify `BUILD_SHARED_LIBS=ON` shouldn't this be `BUILD_SHARED_LIBS=TRUE` ?
Thx!

kdewald, 2023-02-23:
@laborg thanks for the fix! Can you send me the link to your project so that I can send people your way?

laborg, 2023-02-24:
@kdewald As soon as they are released I will tag you. The idea is to create two projects: 
 - `SimpleBLE_jll` which makes the shared library available on Julia (see https://docs.binarybuilder.org/stable/jll/) and
 - `SimpleBLE.jl` a Julia package to offer a higher level abstraction for Bluetooth LE calls

laborg, 2023-02-27:
@kdewald 
The builds for 
 - simpleble_shared_linux-armv6.zip 
 - simpleble_shared_linux-x86.zip 
still include the static libraries. Should CI automatically upload binary artifacts resulting from a merged PR or is there a manual step involved?

kdewald, 2023-02-27:
@laborg There's something broken in the build process that I still need to debug, so those two will be wrong for a while longer while I figure it out.

laborg, 2023-02-28:
Could it be that this is a timing problem with parallel builds?

kdewald, 2023-02-28:
No, this is an architecture problem. CMake tries to link to an incorrect DBus library file. It should be somewhat straightforward to fix.

kdewald, 2023-03-19:
This was fixed in v0.6.1

closed 2023-03-19 by kdewald