simpleble/simpleble #215: Rust bindings issues

kuyoonjo, 2023-03-03:
1. Crash when IOT device disconnected. `set_callback_on_disconnected` not fired.
2. Crash if try to connect a device while scanning.
3 `set_callback_on_connected` never fired.

kdewald, 2023-03-21:
Hey @kuyoonjo, could you please confirm which operating system you are using?

kdewald, 2023-03-21:
Also, I haven't been able to reproduce issues 1 and 3, but did reproduce issue 2.

kdewald, 2024-07-27:
We're going to rewrite the entire Rust backend from scratch, so all of these related issues will be obsolete (and replaced by newer ones).

kdewald, 2025-05-20:
Just finished rewriting the new Rust backend. I'll reopen to keep track of this issue and we'll try to validate that it doesn't happen again.