simpleble/simpleble #218: Missed advertisements

duichan, 2023-03-08:
I am evaluating simpleBLE on a Raspberry Pi 4B (latest Bullseye 32bit OS) with a RuuviTag BLE environment sensor. I know that the Ruuvi is broadcasting a manufacturer-specific data advertisement approximately every 1.25 seconds, but using simpleBLE I am only picking up about every fourth one. The program I am using is your simplebluez/example_scan (except with the scan duration on line 71 increased from 3 s to 100 s).  Am I doing something wrong or is this an inherent limitation of simpleBLE?
Incidentally if I use hcidump from the command line then I can indeed see all the advertisements.
Thank you.

duichan, 2023-03-08:
Wow, thanks for your very quick response. The curious thing is that hcidump on the same system seems to pick up nearly everything, which makes me think that the scan interval/window is somehow different with simpleBLE.

eriklins, 2023-03-08:
I think hcidump - as the name implies - would be more like a sniffer and just dump anything coming in over the BT hci interface without taking a scan interval or window into account. Maybe you can try "hcitool lescan" which might operate on a higher application level and hence use scan interval/window. But no idea whether it would adjust scan interval/window to certain values.

duichan, 2023-03-08:
OK thanks.

closed 2023-03-08 by duichan

kdewald, 2023-03-13:
Hey @duichan, sorry for my late response.

For the missed packets issue that you see, the main cause I've seen is compiling the SimpleBLE in debug mode (which is the default), which severely affects performance. You need to use the `-DCMAKE_BUILD_TYPE=Release` flag to build with all optimizations on.

duichan, 2023-03-14:
Good call! Thank you.