simpleble/simpleble #220: SimpleBLE C and C++ examples failing in Linux with DBus exception error

Dharani3110, 2023-03-10:
I get the below DBus issue when trying to execute examples in **simpleBLE** **c** and **cpp** folders. But **simpledbus** example seems to work.

**My environment:**
**OS:** Ubuntu 22.10 Virtual machine
**Installed packages:** libdbus-1-dev
```
terminate called after throwing an instance of 'SimpleDBus::Exception::DBusException'
what():
org.freedesktop.DBus.Error.MatchRuleNotFound: The given match rule wasn't found and can't be removed
Aborted (core dumped)
```
Please help me in solving this issue.

kdewald, 2023-03-14:
Hey @Dharani3110, can you check if your BLE adapter is supported in the VM? We've had some issues in the past where the OS doesn't really have access to any outside peripherals.

Also, which VM app are you using? VirtualBox? VMWare?

Dharani3110, 2023-03-14:
To answer your question, I use Oracle VirtualBox. Let me check if OS has access to outside peripherals and get back.

Dharani3110, 2023-03-15:
After adding my host bluetooth adaptor to the USB devices of the Ubuntu virtual machine, now I get the "Bluetooth not found" error status instead of the DBus error.

kdewald, 2023-03-16:
It's quite likely that the virtualization layer is failing to properly expose this to Ubuntu. 

My biggest recommendation would be to try this directly on the host, given that it's already supported.

Dharani3110, 2023-03-17:
Sure Thanks!

Dharani3110, 2023-03-17:
Sure, Thanks for your support!

closed 2023-03-17 by Dharani3110